Ar-zz-duboy/Arduboy #11: Add arduboy logo to boot sequence

yyyc514, 2015-05-29:
[image]

Can we agree that (by default) unless a sketch chooses not to compile it (for space reasons) that at boot time we should scroll in the Arduboy logo?  I downsized the graphics from Arduboy.com and rasterized it exactly but if there isn't agreement on this I probably won't go further.

I thought it would be nice to have some consistency and build in a standard "device experience" that would include toggling brightness, sound, boot logo, etc.  Kind of help with the branding of the Arduboy experience.

Another option would be to compile the boot sequence into the Bootloader so that it's completely outside of a sketch... I'm very curious how much space is to spare there.  I know the default on Atmega32U4 is to use 4k for the boot loaded, but I"m not sure how much of that space is actually used by the boot loaded now.

yyyc514, 2015-06-04:
It looks pretty slick.  I have it worked over in https://github.com/yyyc514/ArduboyLib/tree/brightness if anyone wants to try it.  I'll make a sep PR later.  I did add a define so devs could easily turn it off if they really really didn't want it - though I honestly think it should be part of the hardware (bootloader).

yyyc514, 2015-06-10:
Is this welcome in core?  If not I'm going to go ahead and put it in the Extra lib I'm building for Arduboy.  If it's welcome I'll make a PR.

rogosher, 2015-11-07:
Yes, please make a PR for this. The branding is very slick. I agree that if the developer wishes to override the boot logo the option should be easily available.

yyyc514, 2015-11-07:
The difficulty is how to turn it off... it's the kind of thing you'd want to use defines for, but the root sketch file of a project has no way to set defines that are evaluated before the entire project is compiled... and if you make it a programatic option then in my experience C is still going to link it in the final sketch (even if it's never called)... because technically it could have been called.

That means whatever space we spent storing the logo is wasted for anyone not wishing to show the logo... unless they are comfortable editing the library itself, in which case there is no problem.  All the problems go away if we accept most people might not use the core library as-is but rather would download it and then edit one of the H files to set preferences, etc.

rogosher, 2015-11-08:
I think we can use the assumption that most won't mind the loading sequence, and would even prefer it, and like you say: those that would rather not use it can modify the source. I say go ahead and do a PR without the option to turn it off. Kevin was interested in getting this added as well.

yyyc514, 2016-02-04:
Sweet!

MLXXXp, 2016-02-08:
@rogosher wrote:
> I think we can use the assumption that most won't mind the loading sequence, and would even prefer it, and like you say: those that would rather not use it can modify the source.

The problem with disabling the logo using  a *#define* in a library header is that it will affect **all** sketches that use the library. My assumption is that the main reason to turn off the logo is because it's necessary to free up some program space, not because it's an annoyance to someone.

What  happens if someone writes a "killer" game that many people like? Say this game requires so much space that the only way to fit it is to free up space by disabling the logo. If a user disables the logo to play this game, then it will be disabled for all other sketches that this user loads (assuming that the Arduino IDE  or codebender is used, and the *#define* isn't changed back and forth between loads). If the game becomes quite popular, or a number of games have the same issue, then the logo may end up being disabled by a good many people, even though they wouldn't mind seeing it on smaller games.

@yyyc514 wrote:
> The difficulty is how to turn it off... it's the kind of thing you'd want to use defines for, but the root sketch file of a project has no way to set defines that are evaluated before the entire [...]

I can think of one way around this. It's a bit radical and makes library maintenance a bit more difficult, but is fairly straight forward for a sketch developer. Instead of one library, there would be three (or more).

The current *Arduboy* library and its *Arduboy* class would be renamed *ArduboyBase*.

Function *logoIntro()* in the *ArduboyBase* class would be made pure virtual, thus making *ArduboyBase* an abstract class.
In *ArduboyBase.h*:
`virtual void logoIntro() = 0;`
The `#define SHOW_LOGO` and tests for it would be removed and *logoIntro()* would always be called.

A new library and class would be created called *Arduboy*, in its own directory separate from *ArduboyBase*. Class *Arduboy* would be a derived class inheriting *ArduboyBase*. The *Arduboy.h* file would be:
```cpp
#ifndef Arduboy_h
#define Arduboy_h

#include "ArduboyBase.h"

class Arduboy: public ArduboyBase
{
public:
  void logoIntro();
};

#endif
```
*Arduboy.cpp* would contain the code for function *logoIntro()* as it's currently written.

So sketches would remain unchanged, with a `#include "Arduboy.h"` and the logo would be displayed as it currently is.

Now, for sketches that want to gain some space by not showing the logo, we create another library called *ArduboyNoLogo*, again in its own directory. This library still defines a *Arduboy* class (not *ArduboyNoLogo*). The difference is that its version of *logoIntro()* would do nothing, and also any bitmaps wouldn't be included (e.g. *logo[]* in *assets.h*).
In *Arduboy.cpp* there would be `void ArduboyNewLogo::logoIntro() {}`.

A sketch developer would just have to replace `#include "Arduboy.h"` with "ArduboyNoLogo.h" and, viola, no logo is displayed and more code space is available.

This technique makes it possible to easily have alternate logos, as well. If sketches that had lots of free code space wanted a logo that zoomed in, with fireworks around it, and played the *1812 Overture* in 4 part harmony, a library called *ArduboyFancyLogo* could be created and included in place of *Arduboy* :grinning:

MLXXXp, 2016-02-08:
If it turns out that we want to address the possibility that some users may want to turn off the logo because it annoys them or they want start up to be faster, I think there should be a flag reserved in EEPROM to indicate whether the logo should be displayed or not.

The *logoIntro()* function itself would check if a specific button or buttons were pressed when it starts, and toggle the flag if so. It would then display the logo or just return, depending on the flag.

yyyc514, 2016-02-08:
This doesn't need to be done with two classes and having a software time flag doesn't help you because then you don't get any of the space back - it's still there all the time wasting space.

All we need is begin() and beginWithoutLogo() if we want to make the logo super easy to remove on a per-sketch basis... or even if we don't do that it's not hard for sketches to do a custom startup sequence... I was planning on documenting that in the README IF we figure out how to get the logo working at all in a reasonable amount of space.

yyyc514, 2016-02-08:
> If it turns out that we want to address the possibility that some users may want to turn off the logo because it annoys them or they want start up to be faster, I think there should be a flag reserved in EEPROM to indicate whether the logo should be displayed or not.

I'm not opposed to this but I'll leave it up to Kevin.  Honestly I think the branding is important and we should have found a way to put it in the boot ROM so no one can even disable it at all... and it's already going to be easy enough for sketches to turn it off that want to... so I don't know if we need to go all the way and make it a configurable setting based on EEPROM.

I'll ask Kevin.

yyyc514, 2016-02-17:
Closing this, PR for boot logo was just merged.

closed 2016-02-17 by yyyc514