Ar-zz-duboy/Arduboy #14: Incorporate complete USB power saving when not using USB

yyyc514, 2015-06-04:
Ref:
http://harizanov.com/2013/02/power-saving-techniques-on-the-atmega32u4/

Not only do you power USB down you also have to freeze the clock.

rogosher, 2015-11-07:
Perhaps we can create a development branch and include this for some testing. What do you think?

yyyc514, 2015-11-08:
My core lib repository already has code for most of this.

rogosher, 2015-11-08:
Can you link to the relevant code? Thanks.

yyyc514, 2016-01-31:
Actually all the work I've done so far is in the main repository now.

yyyc514, 2016-02-07:
So far, this issue itself is still open though.  USB sucks power from the battery constantly, and that's what this issue was meant to address.

yyyc514, 2016-02-09:
For anyone else who wants to tackle this is what needs to happens:

1. Power down USB fully according to the link above
2. You want to hook the VBUSTI interrupt and when you see it power USB back up.

Then test to see if this actually works and lets you reflash the device. Or we could just turn off USB by default when on battery power and to reflash you have to plug your Arduboy in then turn it on.

The power savings he reports for turning off USB are almost 50% (for the CPU):

```
delay(8000);  // 11.35mA
 
power_usb_disable();
delay(8000);  //9.97mA
 
USBCON |= (1 << FRZCLK);             // Freeze the USB Clock              
PLLCSR &= ~(1 << PLLE);              // Disable the USB Clock (PPL) 
USBCON &=  ~(1 << USBE  );           // Disable the USB  
delay(8000);  // 5.3mA
```