Ar-zz-duboy/Arduboy #15: Publish library to Arduino Library Manager

yyyc514, 2015-06-04:
Reference:

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Looks like just adding the metadata (my PR already does this) and then just asking they add it to their batch job... and tagging official releases (so their job finds them).

rogosher, 2015-11-07:
Yes, thank you. Please add these files. At the very least, the link is extremely helpful.

yyyc514, 2016-02-07:
We need to hurry towards a 1.0 and then tag something to get this moving on their end.  Or at least maybe tag a 0.8 or 0.9?

rogosher, 2016-02-07:
Let's go with 0.9

yyyc514, 2016-02-07:
I'll tag what we have now as 0.8.5 (it fits plus it's nostalgic from when I started Ruby on Rails).  Plan to tag 0.9 in the next day or two.  We should go ahead and reach out to Arduino to get added to the official list though.

rogosher, 2016-02-08:
Alright, I'm caught up to date. I'd like to do this as a github release, instead of the tag method unless there is a reason not to. Right on, we are pretty much ready for the release too. It would be nice to get the boot sequence done.

MLXXXp, 2016-02-08:
One main thing to decide is whether the library should be Arduino 1.0.x backward compatible or should be 1.5.x only. This will determine the directory structure.

One drawback to making it 1.0.x compatible is that when people use *Sketch -> Include Library* in the IDE, it will pull in additional unnecessary headers other than *Arduboy.h*, such as *audio.h* and *assets.h*. If 1.5.x only, this could be avoided by putting the headers in subdirectories.

Conversely, the advantage of 1.0.x compatibility is self explanatory.

yyyc514, 2016-02-09:
> If 1.5.x only, this could be avoided by putting the headers in subdirectories.

Can you confirm that actually works?  I haven't found subdirectories to work.  Link to example lib doing that?

MLXXXp, 2016-02-09:
> I haven't found subdirectories to work.

You have to use relative paths for any *#include*s for files not in the same directory.

For a directory structure such as:
```
├── examples
│   └── Tunes
│       ├── README.md
│       └── Tunes.ino
├── extras
│   ├── CONTRIBUTORS.md
│   ├── Doxyfile
│   ├── LICENSE
│   ├── README.md
│   └── TODO.md
├── keywords.txt
├── library.properties
└── src
    ├── Arduboy.cpp
    ├── Arduboy.h
    └── utility
        ├── audio.cpp
        ├── audio.h
        ├── core.cpp
        ├── core.h
        └── glcdfont.c
```

In *Arduboy.h* or *Arduboy.cpp* you would use:
```cpp
#include "utility/core.h"
#include "utility/audio.h"
#include "utility/glcdfont.c"
```
For *#include*s for files in the same directory you don't need a path, just the file name.

I used the above directory structure with `#include "utility/core.h"` and `#include "utility/audio.h"` in *Arduboy.h*, and `#include "utility/glcdfont.c"` in *Arduboy.cpp*. Sketches compiled properly (using Arduino IDE 1.6.7).

MLXXXp, 2016-02-09:
Note that the subdirectory doesn't have to be named *utility* and you can have more than one. So, for example, we could use *subclasses* for core and audio, and *fonts* for *glcdfont.c*

rogosher, 2016-02-09:
Per the docs, you do not need to match the 1.5 folder layout. 

As a personal preference, I do usually try to separate into a `src` and `include` folder. Files that are related to building though, should be in the root, same with REAMD.me, so github can do it's thing. In fact, all of extra could be left in root. Once the api reference is up, we can yank the `doxyfile` out of the repo and create another repo that serves as the api's build repo.

MLXXXp, 2016-02-09:
> As a personal preference, I do usually try to separate into a src and include folder. Files that are related to building though, should be in the root, same with REAMD.md, so github can do it's thing.

@rogosher, you're right, I should have left the main README.md in the root, for GitHub.

However [the specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification) recommends that a given source file and its associated header file be kept in the same directory, as can be seen in [the complete example](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#a-complete-example).

I'm not sure what "files related to building" there would be in the Arduino world.

yyyc514, 2016-02-10:
> I used the above directory structure with #include "utility/core.h" and #include "utility/audio.h" in Arduboy.h, and #include "utility/glcdfont.c" in Arduboy.cpp. Sketches compiled properly (using Arduino IDE 1.6.7).

I know about relative paths. :-) I've just NEVER seen this work - did something change in 1.6?  Arduino IDE COPIES all your files into the build folder and in the past has not preserved subduers (or it didn't copy those files, so nothing worked).

I think we need to stop here and consider if we care about 1.0.x, because I swear this didn't use to work... that said I'm all for a few subdirs (not too many).  Which I think you already said, right?

I don't see the need for a "subclasses" folder... a folder can have more than 2 files and be ok.

Things like LICENSE, README and CONTRIBUTORS shouldn't be moved from root... that's where I expect to find them on other projects and GITHUB helpful shows the readme and even categorized based on license (I thought).

Lets first figure out if supporting 1.0 is required and then consider what changes to the organization might make sense.

yyyc514, 2016-02-10:
Ok some of that was already said, sorry - didn't finish reading.  I honestly don't find things super messy now, but not opposed to a "src" folder.  At this point the only things I'd really consider for subfolders are perhaps assets.h and fonts.h... the list of files is just not that long.

MLXXXp, 2016-02-10:
> Lets first figure out if supporting 1.0 is required [...]

I'm kind of conflicted about this but leaning towards 1.5+ support only (actually 1.6+ since the 1.5 series was just beta).

1.6.x offers many enhancements over 1.0.x, both from an IDE usability standpoint and for improved libraries and additional functions. Just one example: The [EEPROM library](https://www.arduino.cc/en/Reference/EEPROM) is greatly enhanced and, among many other things, now has an [*update()*](https://www.arduino.cc/en/Reference/EEPROMUpdate) function that only writes if the value has changed, thus possibly extending the limited life of the EEPROM.

Another downside of using 1.0.x is that sketches have to *#include* all libraries that the desired libraries are dependent on. This is no longer necessary with 1.6.x

My main reason for supporting 1.0.x is from the point of being a Linux user, specifically Ubuntu. The latest Arduino IDE version that you get when you install from the package manager (software center) is 1.0.5 and it appears that this will also be true for Ubuntu long term support version 16.04, coming in April. It's a bit complicated to install the latest Arduino version by bypassing the Ubuntu package manager. However, it's likely that the number of Linux users will be minimal, and many of those will be comfortable with the installation process.

For Windows and probably Mac, I don't think insisting on 1.6.x or higher would be a problem.

MLXXXp, 2016-02-10:
On thing to consider; I don't know if Kevin is still pursuing getting [*Arduino at Heart*](https://www.arduino.cc/en/ArduinoAtHeart/HomePage) certification, but if so, following the [*library specification*](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification) recommendations as closely as possible may be a requirement, or at least beneficial.

rogosher, 2016-02-10:
MLXXXp, yes your PR is right on track with that. And even though the loose requirements for the library don't require matching the lib spec that closely, your suggestion that we should follow it closely is probably the right one.

yyyc514, 2016-02-11:
Is there no apt repository for easily getting a newer version on existing Ubuntu popular releases?

Thanks,
Josh

MLXXXp, 2016-02-11:
>Is there no apt repository for easily getting a newer version on existing Ubuntu popular releases?

I searched for quite a while and didn't find anyone providing a [PPA](https://help.launchpad.net/Packaging/PPA) for it.

I ended up using [this site](http://ubuntuhandbook.org/index.php/2015/11/install-arduino-ide-1-6-6-ubuntu/) as a guide to install 1.6.7. It worked quite well but it caused some issues with having the 1.0.5 version, from the Ubuntu repository, installed at the same time, mainly with the icon on the launchpad. But, I can used either one with just minor annoyances.

I wanted to keep 1.0.5 around in case I want to see what's different or to test things for backwards compatibility, but eventually I'll probably remove it.

yyyc514, 2016-02-12:
Ha, `guide`... That's just "install the great binary distribution we provide in opt"... so really for Linux they provide a great option for anyone familiar at all with Linux.  I don't see a need to support 1.0.5 in that case.

MLXXXp, 2016-02-12:
>That's just "install the great binary distribution we provide in opt"...

Yup, but it pointed out the `./install.sh` command that easily integrates it with the desktop.

Most other articles I found just said "run the *arduino* command" to bring up the IDE. Not something most users would want to do.

yyyc514, 2016-02-12:
That's honestly pretty standard fare for Linux.  Tar/gz provided binary with shell script to do some sort of install... it's the equivalent of manually downloading a zip on Mac and copying the icon to Applications instead of getting it from the App Store.  If that's scary you should maybe not be on Linux. :-)

MLXXXp, 2016-02-12:
>That's honestly pretty standard fare for Linux.

In general, that's true, but not so much for Ubuntu. Ubuntu is one of the "go to" distributions for less techinical users just wanting an alternative to Windows or Mac. They might only be comfortable with using the GUI desktop and installing applications using the graphical Software Center, and for the most part, it allows that.

>If that's scary you should maybe not be on Linux. :-)

But that's a main goal of Ubuntu; to make Linux not so scary, and easy to use for the general population.

It's not scary for me. I've been using Linux, and computers in general, since before GUIs were available. (My first computer was 8080 based with 256 **bytes** of RAM, built from a kit in 1977). For a long time, the primary Linux distribution I used was [Gentoo](https://www.gentoo.org/), a source based system. My main profession is writing software/firmware for embedded systems, many of them Linux based. I now mainly use Ubuntu because it mostly "just works" and I don't want to spend a lot of time fiddling with it instead of using it. I still make heavy use of the command line, though :wink:

yyyc514, 2016-02-12:
I know what Ubuntu is, still feel it's a bad choice for "the masses".  Linux on the desktop has sadly never arrived, despite what same say.  If you want a *nix for normal people, get a Mac. :-)  And statistically, given current trends, you might... given time... every year PCs are down and Macs are up. :-)

I'll never go back. I used Linux and BSD for years before I found OS X.

MLXXXp, 2016-02-12:
>I know what Ubuntu is, still feel it's a bad choice for "the masses".

Maybe true for "the masses" but it's the kind of thing someone technical enough to want to learn coding, via the Arduboy, might have decided to use.

Anyway, I think we're in agreement. Possible Linux issues shouldn't deter us from making the library compatible with IDE 1.5+ only.

>I used Linux and BSD for years before I found OS X.

I used OS/2 before switching to Linux :grin:

rogosher, 2016-02-14:
I think you are safer looking at a linux deployment in terms of the package manager. Having `apt-get` or `yum` support is the target you want to hit when trying to make this easy for a user and hit all the big distros. It seems the common practices are to append a line to `/etc/apt/sources.list`, or package a `.deb` and `.rpm` file while providing links static download links.

Examples
- https://docs.docker.com/engine/installation/
- https://www.vagrantup.com/downloads.html

MLXXXp, 2016-02-14:
>I think you are safer looking at a linux deployment in terms of the package manager.

Of course, but no one has done this for the Arduino 1.6.x series, that I'm aware of. Care to volunteer?

rogosher, 2016-02-16:
> > I think you are safer looking at a linux deployment in terms of the package manager.
> Of course, but no one has done this for the Arduino 1.6.x series, that I'm aware of. Care to volunteer?

I can at least start the research on this and provide some best practices. If it's possible to push a nice release through a script, heck yeah, I'll work up something. Maybe we can talk about it a bit in IRC, do you use it?

MLXXXp, 2016-02-16:
>Maybe we can talk about it a bit in IRC, would you care to join the channel? There are other options for chatting as well.

How about starting a topic in the [Arduboy community forum](http://community.arduboy.com/)?

MLXXXp, 2016-02-16:
Has a decision been made whether to maintain 1.0.x compatibility or not? If not, I think the source should be moved to a *src* directory.

And, as I've already pointed out, if we don't want
```cpp
#include <audio.h>
#include <core.h>
```
to be added when *Sketch -> Include Library -> Arduboy* is used in the IDE, then *audio.h*, *audio.cpp*, *core.h*, *core.cpp* should be moved to one or more subdirectories under *src*. The same likely goes for any other *.h* files that shouldn't be *#include*d in a sketch, plus their assodiated *.c* and *.cpp* files.

rogosher, 2016-02-16:
Can we get a PR for the source before 1.0.0's release? If so, let's do the change.

MLXXXp, 2016-02-16:
Is one subdirectory sufficient?

Do we want to name it *utility* as the [Library Specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#source-code) uses as an example, or some other name?

*library.properties* and *keywords.txt* must remain in the root. *README.md* and *LICENSE* probably should as well. However, should *CONTRIBUTORS.md* and *TODO.md* be moved to *extras*, to follow the specification recommendation more closely?

Should we even have a *TODO.md*? Wouldn't using *issues* under GitHub suffice? It's one more thing that must be maintained and kept in sync.

rogosher, 2016-02-16:
I was debating the `TODO.md` this morning. I think it does give anyone downloading the library a set of goal they can work on. It does need to be updated. I would like to avoid an `extras` folder if possible and focuse on getting a `src` folder, and an `include` (or whatever is best/recommended in this case). Leave github/arduino stuff in the root, as well as project files.

MLXXXp, 2016-02-16:
>focuse on getting a `src` folder, and an `include` (or whatever is best/recommended in this case).

I don't think we should separate *.h* files from their same-named *.cpp* files. It's not the Arduino recommended way and, IMHO makes them harder to find. Therefore, you probably don't want a subdirectory named *include*.

But if you don't want `#include core.h` and `#include audio.h` (and `#include ab_printer.h` if we break out printing) to show up in the user's sketch when they use *Sketch -> Include Library -> Arduboy*, then *core.h*, *core.cpp*, *audio.h*, *audio.cpp*, *ab_printer.h* and *ab_printer.cpp* need to go in a subdirectory under `src`.

To quote the specification (emphasis mine):
>in general, the only header files in the root src/ folder should be those that you want to **expose to the user's sketch** and plan to maintain compatibility with in future versions of the library

So my question is, should a subdirectory under *src* that contains *core.h*, *core.cpp*, *audio.h*, *audio.cpp*, *ab_printer.h* and *ab_printer.cpp*, etc. be called *utility*? Is there a more desirable name?

>I would like to avoid an `extras` folder

Although it doesn't seem to strictly follow the recommendations, not using *extras*, and leaving *CONTRIBUTORS.md* and *TODO.md*, in the root is fine with me.

rogosher, 2016-02-16:
Alright, well, you do know the recommendations a lot better than I do. Go ahead and submit something you think is fully in line with the spec, you would probably know best here.

MLXXXp, 2016-02-16:
The other way to go would be to have a separate subdirectory for each subclass and name them accordingly. So we would have:

```
└── src
    ├── Arduboy.cpp
    ├── Arduboy.h
    ├── glcdfont.c
    ├── audio
    │   ├── audio.cpp
    │   └── audio.h
    ├── core
    │   ├── core.cpp
    │   └── core.h
    └── ab_printer
        ├── ab_printer.cpp
        └── ab_printer.h
```

MLXXXp, 2016-02-16:
Pull request #91 created as per my previous comment.

yyyc514, 2016-02-17:
Can this be submitted to Arduino now to get included in their official list so people with the IDE can easily download the lib?

rogosher, 2016-02-20:
I've created a release and open an issue, found here https://github.com/arduino/Arduino/issues/4597.
Looking at the closed issues, they will let us know if there is a problem.

yyyc514, 2016-02-20:
1.1 should not be marked released if this item isn't 100% complete.

rogosher, 2016-02-20:
Closing this issue now that it has been pushed. If any feedback is given in the issue I will create a new issue. Thanks @yyyc514!

closed 2016-02-20 by rogosher