Ar-zz-duboy/Arduboy #16: Arduino IDE boards.txt file and instructions

yyyc514, 2015-06-04:
This is needed:

* So Arduino users don't have to choose "Leonardo" all the time... they can choose Arduboy.  
* If we're going to allow 16 and 8Mhz compilation choices.  
* It will be needed in the production version anyways to properly set the USB vendor and product ID.

Reference:
https://github.com/yyyc514/hardware_arduboy

This might be in another reposition Arduboy hosts, but since we just have the one now I'm filing this issue here.

rogosher, 2015-11-07:
Yes. At the very least let's set the board option. The other two options may be needed later.

yyyc514, 2016-02-09:
https://github.com/yyyc514/hardware_arduboy

I removed the speeds and just added devkit and production.  About to go fix my PR to work with this.  I think Arduboy should fork this repository and maintain this.

yyyc514, 2016-02-09:
[image]

yyyc514, 2016-02-09:
We still need to make sure the boot fuses are current (if someone every wired up the ISCP ports).  Although right now I'm getting a warning that it can't even find the boot loader (it's looking locally not in the Arduino source), so maybe that's not a huge issue?

Can someone else test this?

yyyc514, 2016-02-09:
Or we could call it "Kickstarter Edition" or any number of other cute things.

yyyc514, 2016-02-09:
We probably want to get on this list:

https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls#list-of-3rd-party-boards-support-urls

We can publish a JSON file so someone just types the URL into the Arduino IDE and then it does the actual download the the boards file including keeping it up the date.  Worth pursueing?

rogosher, 2016-02-09:
Yes, I would say say this is worth keeping as an active issue. Anything that we can do on this end to create a better user experience is going to be a win. I do not have the ability to test this at the moment.

yyyc514, 2016-02-10:
I've built for both devkit and production now using the IDE - seems to work great.

yyyc514, 2016-02-10:
Yikes.  https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

That is one hell of a JSON payload (and of course ours would have fewer boards), but not impossible.

yyyc514, 2016-02-13:
Should we clone this if we agree it has value and include it in the 1.0 release notes?

rogosher, 2016-02-14:
Yes

yyyc514, 2016-02-14:
Please fork then, and we can maintain here.

rogosher, 2016-02-16:
Alright, I simply don't have a production unit to test on. Opinions? Push this back to a 1.x milestone?

MLXXXp, 2016-02-16:
A haven't looked into the details of what's being discussed here, but whatever [SparkFun does for their boards](https://github.com/sparkfun/Arduino_Boards) works well for me. The details in their *README.md* make it quite easy for the user.

yyyc514, 2016-02-17:
It works on production and devkit. I've tested both.

yyyc514, 2016-02-17:
Yeah the readme and such things could probably be refined.  I just wanted to get the core experience working and leave enough for someone who can follow instructions to get started.

If we want to polish it up a lot more that's fine by me.  I never cared to really own this, just at the time no one else seemed interested in a proper boards file.

yyyc514, 2016-02-17:
I went ahead and tagged 1.0 since all the PR related blockers are cleared, but this is really pretty important for someone testing their work on both production and the devkit.   Lets fork this and update the docs if need be... is there really anyone against that course of action?

rogosher, 2016-02-17:
Yeah, I don't want to include this feature if it hasn't been fully tested. I do own this thing, as I have a dev kit. I might expedite myself a production unit tomorrow and get the production testing by another set of eyes done.

yyyc514, 2016-02-17:
It seems wise for you to have a production kit if you're the person in charge of items like these.  I'm not sure what "include" here means though... does it mean "Arduboy" can't ever have any experimental repositories?  That seems needlessly restrictive. *shrugs*

Just wanting this off my plate and feeling like once it's in Arduboy someone else can take care of it.  Source code is far more interesting than board files.

rogosher, 2016-02-17:
I pushed to get you a production unit ASAP after the first few were ready. I did not push to get myself one. Now there is stock available again and I will have have a unit. :100:

yyyc514, 2016-02-17:
Well, thanks.

MLXXXp, 2016-02-17:
I'll have a "production" unit equivalent once I rewire my home made version. I then won't have a "DevKit" equivalent any more, though. :wink:

yyyc514, 2016-02-21:
This needs to happen.  It really really bugs me reading the readme that people have to pick Leonardo when I did this work like 6 months ago.

MLXXXp, 2016-02-21:
It's more of a problem that you can't chose between a production or DevKit unit, to automatically set the proper *#define*s for the library.

yyyc514, 2016-02-22:
Right. That is the other big win. Thought defaulting to produciton was no doubt the right choose for a lot of new users receiving their units just now.

MLXXXp, 2016-02-22:
>It really really bugs me reading the readme that people have to pick Leonardo when I did this work like 6 months ago.

It bugs me that the production Arduboy comes with the Leonardo bootloader and thus the Leonardo USB IDs. It should have had its own USB IDs. I doubt the Arduino people would be very happy about this, either.

yyyc514, 2016-02-23:
Well, it's not possible to fix that now, but it is possible to fix getting this out there for the world to benefit. :-)

rogosher, 2016-03-25:
> It bugs me that the production Arduboy comes with the Leonardo bootloader and thus the Leonardo USB IDs. It should have had its own USB IDs. I doubt the Arduino people would be very happy about this, either.

I have about the same feeling, but it something we on the development end don't have control over.

rogosher, 2016-05-27:
Arduboy hardware target added https://github.com/Arduboy/arduboy.github.io/tree/master/board-support

closed 2016-05-27 by rogosher