Ar-zz-duboy/Arduboy #18: Figure out how to remove drawRect calls from drawChar

yyyc514, 2015-06-11:
These get compiled in and waste space for anyone not using enlarged font sizes. This uses 216 bytes extra just to compile in drawRect and all it's dependencies when it's never used.

Not sure about the best fix here, but open to suggestion.

yyyc514, 2015-06-11:
Funny as it looks you can save 100+ bytes by removing all the special casing between drawing different sizes:

```
    for (int8_t j = 0; j<8; j++)
    {
      draw_color = (line & 0x1) ? color : bg;

      if (draw_color || draw_bg) {
        for (uint8_t a=0; a < size; a++ )
          for (uint8_t b=0; b < size; b++ )
            drawPixel(x+(i*size)+a, y+(j*size)+a, draw_color);
      }

      line >>= 1;
    }
```

yyyc514, 2015-06-30:
Objections?

yyyc514, 2015-11-07:
Ping.

rogosher, 2015-11-08:
How does it change the look between font sizes?

yyyc514, 2016-01-31:
It doesn't change the look.

yyyc514, 2016-02-06:
Creating PR, closing.

closed 2016-02-06 by yyyc514