Ar-zz-duboy/Arduboy #20: Clarify what blank() is for or remove it entirely

yyyc514, 2015-06-11:
If we want to just turn the screen on and off we could probably do that at the hardware level.  For a buffer backed game I"m not super sure what the advantage of blanking the screen is with non blank data in the buffer.  I think `blank()` could easily be confused with `clearDisplay()`.

yyyc514, 2015-06-11:
I think I'd suggest replacing blank() with `screenOn()` and `screenOff()` or similar, so the function - and how it's different than clearDisplay() is clear.

yyyc514, 2015-06-19:
This makes more sense in a low-level library (#32).  I pulled it into ArduboyCore since when there is ONLY the screen (no buffer) it loses it's ambiguity.  Blank means blank.  I think perhaps if we keep it we should consider just mapping it to clearDisplay()... so whichever one you call results in blanking the buffer when it's next pushed to the display.

yyyc514, 2015-11-07:
Ping.

rogosher, 2015-11-08:
I think renaming it is a good idea unless there is a convention to use `blank()` that I'm not aware of. Keeping `blank()`, mapping it, and depreciating it now with removal in the next major version would be a good idea.

yyyc514, 2016-02-06:
I think we rename as follows:

```
clearDisplay -> clearBuffer()
blank -> blankDisplay()
```

And deprecate the old functions.  Thoughts?

yyyc514, 2016-02-07:
Renamed clearDisplay to clear() and left blank alone since really it's a hardware function in core.  The docs already explain the difference.  Save the rest of this battle for another day.

Deprecated clearDisplay().  PR done.

closed 2016-02-07 by yyyc514