Ar-zz-duboy/Arduboy #24: Add UI for safe mode

yyyc514, 2015-06-12:
Any objection to including a bitmap image to show so someone knows they are in safe mode?  Blinking the LED might be another option... in some cases you might be using safe mode to recover from a sketch that hasn't drawn anything to the screen, so it seems nice if it had some type of UI... but at the same time I don't want to REQUIRE linking in the font table or even drawBitmap.  Those are big data sinks if someone isn't otherwise using them.  Expensive to require them just to say "Safe" in safe mode... For a bitmap of "Safe" or "Safe mode" you're talking 24-54 bytes.  

Though I guess you could get it as little as low as 15 bytes if you used a 3-4px font.

If production ships with a reset button we might change safe mode so it only is compiled for devkits - and prefer the reset button in production units.

ace-dent, 2015-06-17:
Would prefer minimal resource overhead. Perhaps blink LED 3x times?

yyyc514, 2015-06-17:
I might let this one set a while... I'm thinking once we have production units safe mode might be a LOT less of a big deal.

yyyc514, 2016-02-04:
Going to close this since production has a real reset button... people with dev kits can live without a pretty UI.  if someone else wants to run with this and blink the LED that'd be fine, but I have enough other stuff to do and this has languished.

closed 2016-02-04 by yyyc514