Ar-zz-duboy/Arduboy #27: Bump SPI clock speed to /2 vs /4

yyyc514, 2015-06-16:
This is done like so:

```
SPI.setClockDivider(SPI_CLOCK_DIV2);
```

I'm thinking this should be the default in the core lib.  It reduces the wait time spent doing nothing but waiting on the SPI bus to repaint the screen.  If the final production version is going to run at 8Mhz we definitely want to do this - just to preserve the current SPI speed.  16/4 = 8/2.

I don't have hard #s right this second but in my testing this reduced CPU % for a blank render loop by around 5-10%.  Surprised?  I was surprised that a blank render loop is so slow.  If all you do is use nextFrame at 60FPS and call `clear; print(cpuLoad()); display` out of the box you're talking 31% CPU usage.

With this change and a few others I got that down to 18%.

Thoughts, objections?  If this is good I can do the work and submit a PR.

yyyc514, 2015-06-16:
This is also super important if anyone wants to further hack on grayscale stuff... there is a huge diff in vsync if you are at /2 or /4 SPI clock speed.

yyyc514, 2015-06-19:
Timing a call to `display()` with micros():

SPI /4 SPI.transfer(): 3700 micros
SPI /2 SPI.transfer(): 2670 micros
SPI /2 hand-optimized loop (tight as it can get): 1170 micros

This might have implications for greyscale.  This is over twice as fast as the previous speed I played with.

yyyc514, 2015-06-19:
@ace-dent You asked for hard numbers, there ya go.

rogosher, 2015-11-07:
This would be a good PR for a development branch.

yyyc514, 2015-11-07:
This compiles and works perfectly. I'm not sure why we wouldn't merge it.

yyyc514, 2015-11-08:
So looks like this already got merged.

closed 2015-11-08 by yyyc514