Ar-zz-duboy/Arduboy #30: Issue: write/drawChar/drawRect/font are linked whether used or not

yyyc514, 2015-06-18:
Reference:
https://github.com/arduino/Arduino/issues/3363

This is because write is a virtual function of Print and the compiler can't (or isn't being told) to figure out whether it's actually used or not.  This results in a graphics only game being 1.5-2k larger than it would need to be if it doesn't use any print functions.

yyyc514, 2015-06-19:
The work I've started over in #32 is one possible solution to this.  We could just provide a subclass that was only for graphics (not a subclass of Print)... then none of the print/font stuff would ever get linked.  Unfortunately this isn't possible to build on top of say `Adafruit_GFX` since they insist on subclassing on Print (same as we do now) since that's a more flexible setup - at the expense of 2.xkb of flash.

davidperrenoud, 2015-12-03:
Continuing on @yyyc514's idea, I have created an ArduboyGraphics class, devoid of any text drawing and Print dependency:
https://github.com/davidperrenoud/Arduboy/tree/remove-print-dependency

Developers who do not use text (like @JO3RI) would therefore gain 2 464 bytes of flash.

rogosher, 2015-12-04:
It would be great if we could get some feedback on this issue from people using this branch.

yyyc514, 2016-02-07:
I think one problem with this approach is linking with other libraries that support Arduboy... I don't know enough about C (too used to Ruby's great flexible support of classes)... if I write some awesome libraries for Arduboy (as I have been doing) and then someone comes along and tries to pass me a ArduboyGraphics when I want an ArduBoy or a Arduboy when I want an ArduboyGraphics does the whole world blow up?

Can someone with more knowledge of C weigh in here?  That's why I didn't jump right on this myself - worried there are perhaps good reasons not to do this.  Needs more discussion

yyyc514, 2016-02-07:
One thought I had today is that it might be better to rip text out of Arduboy proper and just add it back via it's own class.  You'd either use that class separately or we could hing into into Arduboy as a variable:

```
Arduboy arduboy;
ArduboyText at (arduboy.getBuffer());
// or
ArduboyText at;
at.begin(arduboy.getBuffer());
at.print("Hi");
// or
arduboy.text.print("hey, what up");
```

In the latter example text would just be an instance of ArduboyText that Arduboy setup for you.

yyyc514, 2016-02-07:
Again I believe we're talking about 2,000 bytes of savings here for programs that are exclusively using graphics and have no intention of rendering text... since it throws out the font, the character drawing, all the print routes, etc.

rogosher, 2016-02-07:
I think you are on the right track here. In order to give the flexibility to developers that allows them to get what they can from the platform, being able to control the stack size in relation to application content is going to be that step.

yyyc514, 2016-02-12:
Is it possible to have Arduboy reference Printing and Printing also reference Arduboy?  I'm stuck.  Printing needs Arduboy to get access to the buffer, drawing, etc... but Arduboy also needs Printer since I'd like it to hold a variable reference.

I want something like:

```
// constructor
    AbPrinter(Arduboy *ab);

// inside Arduboy.h
AbPrinter text(this);
```

But I also want it to be lazy linked in that if text is never referenced in your program that AbPrinter is just junked completely and not linked at all.  Is this approach doomed to failure and it just needs to be an entirely separate class?

JO3RI, 2016-02-12:
on a side note:
What ever approach is going to be followed, it should NOT eat bytes it not used ! Even for things like score (we do score with sprite reference, so we don't need a font), non of our games have a font, nor text as a function.

yyyc514, 2016-02-12:
Right, that's the point.  This is our last chance (other than tunes) to get back a substantial portion of flash.  After this it's all micro optimizations.

yyyc514, 2016-02-12:
If no one answers I'm just going to call it AbText and make it an external variable you have to use in additional to Arduboy so the setup looks like:

```
Arduboy arduboy;
AbText text(&arduboy);

text.print(F("Hello world!")).
```

JO3RI, 2016-02-12:
As a developer, I find your suggestion to be the correct solution.

on an other side note:
We can have optimizations on two fronts ?
- speed
- memory usages (SRAM, RAM, EEPROM)
Once memory usages has been optimized, speed should be also considered. I do realize both can conflict or go hand in hand. (Also, once core is done, each extra library can go live it's own live of optimization)

yyyc514, 2016-02-12:
We're not particularly wasteful with RAM other than the buffer and tunes.  Not much interested in font rendering speed, but if someone else write some awesome code PR is welcome.

yyyc514, 2016-02-12:
Closing and creating PR:

https://github.com/Arduboy/Arduboy/pull/79

closed 2016-02-12 by yyyc514