Ar-zz-duboy/Arduboy #33: Consider subclassing Arduboy from Adafruit_GFX

yyyc514, 2015-06-19:
See my comments about this over in #32.  Unless we're going to hand-tune more than a few of these methods - and IF our API design is going to be pretty much exactly the same, why not just subclass them?  What do we get my having to copy and maintain all those functions individually (and manually port upstream bug fixes, etc).

Obviously we have to implement out own drawPixel (it's a virtual function), and it behooves us to keep using our hardware optimized drawImage (or a better version)... but we're just providing a few low level primitives I'm not sure I see the advantage of having all that additional code to maintain outside of the library it originally came from.

yyyc514, 2015-06-19:
The big issue I see with this now is I'd like to limit our requirement of Print being forced on you.  That's easy to do ourselves, but Adafruit's lib also depends on Print.  I've filed an issue on their Github to see if they are open to having a class without that requirement.

https://github.com/adafruit/Adafruit-GFX-Library/issues/50

yyyc514, 2015-06-19:
Ok pretty sure we don't want to do this unless they remove their use of virtual.  Just a speed trap.

https://github.com/adafruit/Adafruit-GFX-Library/issues/51

yyyc514, 2015-06-19:
Though I suppose we could vendor it in our own repository and rip out the virtual functions... that might be less maintenance work than having our own copy of every single function.

rogosher, 2015-11-07:
Reopened because this might be worth implementing and produce a better code base.

yyyc514, 2016-02-07:
I actually think now this is a pretty bad idea.  It would reduce the scope of our library, but at the cost of size.  ADAFruit is so very, very abstract and fully-featured.  Is small size is a goal I think this would totally sink that ship and prevent us from doing a lot of things we've been doing that are making big wins when it comes to space.

If someone wanted to write a second library that supported ADAFruitGFX it probably wouldn't be that difficult (you have to code some hardware init stuff and drawPixel), but I'm not sure the advantage there - other than maybe portability... but I think a lot of our functions already follow that naming convention, so that might not be as hard as you'd think.

You still want to leave this open and lingering @rogosher

rogosher, 2016-02-08:
It's moved out of the scope of relevant conversation for this repository, I'm going to go ahead and close this.

closed 2016-02-08 by rogosher