Ar-zz-duboy/Arduboy #49: Remove void ArduboyTunes::delay (unsigned duration) {

yyyc514, 2016-01-31:
Can anyone tell me why we need this?  I'm thinking it's a carry over from the tunes library we borrowed from in case you reused timer 0, but we already have the system delay() and timer 0 tracking time so I'm not sure why we need a second delay at all... it doesn't seem to be used internally in the audio class.

Any objections to snipping it?

yyyc514, 2016-01-31:
From the original: (https://github.com/LenShustek/arduino-playtune/blob/master/Playtune.cpp)

```
  // We provide this because using timer 0 breaks the Arduino delay() function.
    // We compute the toggle count based on whatever frequency the timer used for
    // score waits is running at.  If the frequency of that timer changes, the
    // toggle count will be adjusted by the interrupt routine.
```

Yeah this can go.

rogosher, 2016-02-03:
At least there was a comment?

yyyc514, 2016-02-06:
Closing as there is now a PR.

closed 2016-02-06 by yyyc514