Ar-zz-duboy/Arduboy #51: not_pressed() should be notPressed() and start() should be begin()

MLXXXp, 2016-02-04:
Currently there is a function named *not_pressed()*, for testing the state of pressed buttons. In keeping with the [Arduino Style Guide for Writing Libraries](https://www.arduino.cc/en/Reference/APIStyleGuide), which recommends using [Camelcase](https://en.wikipedia.org/wiki/CamelCase) instead of underscores in function names, this should be changed to *notPressed()*.

Hopefully not too many sketches have used this function yet, so it's best to change it as soon as possible to minimise the impact.

The style guide also states:

> Use begin() to initialize a library instance, usually with some settings. Use end() to stop it.

so *start()* should be *begin()*. This one would be harder to change since practically every existing sketch will be using it. Perhaps we could add *begin()* as a synonym of *start()*. All new sketches should use *begin()* and *start()* could be depreciated once all old sketches had been changed.

MLXXXp, 2016-02-05:
If *not_pressed()* does get changed to *notPressed()* it should be added to *keywords.txt*. I didn't put *not_pressed()* in *keywords.txt* on the assumption that it would be changed.

yyyc514, 2016-02-05:
We can change to begin() now and just tag start with the depreciated attribute (I think)...  notPressed is good change, and more C like I think... I probably did that wrong because I'm thinking in Ruby speak, where underscores is more common. ;-)

Though actually perhaps we should just make someone do `!pressed()` and remove `notPressed` completely.

MLXXXp, 2016-02-05:
> Though actually perhaps we should just make someone do !pressed() and remove notPressed completely.

The way the functions are written, *!pressed()* is not equivalent to *notPressed()*. When more than one button is specified, they each serve a different useful purpose. *pressed()* checks if the buttons specified in the argument, as a bit mask, are **all** currently **pressed**. *notPressed()* checks if the buttons specified in the argument are **all** currently **released**.

`notPressed(LEFT_BUTTON + UP_BUTTON)` will only return *true* if **both** the *left* button and the *up* button are not being pressed. I.e. All of the specified buttons must be in a released state.

`!pressed(LEFT_BUTTON + UP_BUTTON)` will return *true* if **either** the *left* button **or** the *up* button is not being pressed (or both). I.e. Only one of the specified buttons (at least) must be in a released state.

MLXXXp, 2016-02-05:
> We can change to begin() now and just tag start with the depreciated attribute (I think)

Yes, it looks like we can use a *depricated* (not *depreciated*; my bad) attribute on the *start()* function. Although it just generates a warning, which will likely not be seen or be ignored by most sketch developers, it can't hurt.

In *Arduboy.h*, I changed
```cpp
  void start();
```
to
```cpp
  void start() __attribute__ ((deprecated));
```
This produced a warning, when a sketch was compiled, similar to:

    test.ino: In function 'void setup()':
    test.ino:8:14: warning: 'void Arduboy::start()' is deprecated (declared at Arduino/libraries/Arduboy/Arduboy.h:48) [-Wdeprecated-declarations]
       aboy.start();
                  ^
So we could change *start()* to *begin()* then add the *deprecated* attribute to *start()* and have *start()* just call *begin()*.

Note: I tried using the *alias* attribute to make *start()* an alias of *begin()* but I couldn't get it to work. If it can be done, the syntax escapes me.

yyyc514, 2016-02-06:
. The way the functions are written, !pressed() is not equivalent to notPressed().

I know, I wrote them, but is this super useful behavior that people couldn't just code themselves?

yyyc514, 2016-02-06:
> So we could change start() to begin() then add the deprecated attribute to start() and have start() just call begin().

Yes, deprecated start would just call begin.  Exactly.

MLXXXp, 2016-02-06:
> I know, I wrote them, but is this super useful behavior that people couldn't just code themselves?

Are you asking if we should keep *pressed()* but remove *notPressed()*? Unless you think we should eliminate both and have developers just implement the desired equivalents themselves using *getInput()*, then I say keep them both.

I used both *pressed()* and *notPressed()* in my [ArduboyLife](https://github.com/MLXXXp/ArduboyLife) sketch. *notPressed()* was used to wait for a pair of keys to both be released. In my case, I could probably have used `getInput() == 0` but that wouldn't have worked if I'd wanted to ignore other buttons that were still held, and I think *notPressed()* reads a bit better if *pressed()* has previously been used.

yyyc514, 2016-02-06:
Ok, lets just rename it then.

yyyc514, 2016-02-07:
See my example: https://github.com/Arduboy/Arduboy/pull/60

yyyc514, 2016-02-07:
Do you think begin needs to extend to the helper classes like audio?  Currently uses `setup`.  If so lets changes that also.  That should (at this point) really just be an internal only change so we don't need to deprecate the old call.

MLXXXp, 2016-02-07:
What about *ArduboyCore::setup()*? Is it needed? I don't think it's used.

yyyc514, 2016-02-07:
No, the idea was you'd use that if you wanted to use core all on it's own, but it's probably way more likely you'd be subclassing from core and just calling boot()... and core is really super low-level, so I don't think it needs a begin - I don't think we should encourage people to use it that way.

So yeah I think we could remove setup() if you want.

yyyc514, 2016-02-09:
Was hoping to roll 0.9 today - think you'll have anytime to get to this?

MLXXXp, 2016-02-09:
I'm planning to do it in the next few hours.

MLXXXp, 2016-02-09:
Pull request #67 created. Closing this issue.

closed 2016-02-09 by MLXXXp