Ar-zz-duboy/Arduboy #59: [space] Change all height, width arguments to int8_t instead of int16_t

yyyc514, 2016-02-07:
Any objections?  This would probably have a significant space savings for programs using lot on the draw functions.  I purposely decide a while back to leave X and Y as 16 bit but the screen is only 64 pixels tall and 128 wide... if there a reason someone would be drawing a image (or object) larger than 255x255 to our 128x64 screen?

For the rare person doing this I think it would be beneficial for them to rethink their code and for us to claim these savings.  Thoughts?

rogosher, 2016-02-07:
I can't conceive of a reason not to implement this, however, maybe someone can come up with a reason to keep the original implementation.

yyyc514, 2016-02-07:
Not huge, but it adds up.  44 bytes saved with Breakout sketch.

yyyc514, 2016-02-10:
Any further thoughts?

yyyc514, 2016-02-10:
Submitted PR for further discussion:

https://github.com/Arduboy/Arduboy/pull/73

closed 2016-02-10 by yyyc514