Ar-zz-duboy/Arduboy #62: Update library to have compile flags for both DEVKIT and RELEASE10

yyyc514, 2016-02-07:
As soon as I get my prototype I'll make these changes.  @rogosher can you confirm for sure the new pin assignments so I'm good to go as soon as I have the hardware?

Per Chris we make production the default option and you have to change the library to define DEVKIT if that's your intention (and I'll also update my board files - and we should considering forking that repository here).

yyyc514, 2016-02-07:
AFAIK it's just the RST, CS, DC, buttons, and speaker pins that changed - the rest of the hardware is unchanged,yes?

rogosher, 2016-02-08:
I think think you are correct there, we could probably start the PR then they can check it and merge or ok it. Chris and Kevin said they were getting together on the 8th to do the github thing.

MLXXXp, 2016-02-08:
>the rest of the hardware is unchanged,yes?

It would be nice if there were one or more functions for controlling the RGB LED (available only if compiled for the production version). I believe the drive to the LEDs is reversed (a high is off and a low is on) so isolating this fact from the user would avoid confusion. I suggest a function like:
`void setRgbLed(uint8_t red, uint8_t green, uint8_t blue)`
where a 0 is off and 255 is full intensity. If desired, the function could be overloaded with other argument styles for specifying the color in different ways.

We might also want function(s) to get the current LED settings.

Also, perhaps ways to set each LED fully on or fully off by just setting the pins high or low instead of using PWM, which might generate faster and/or smaller code. We could probably overload *setRgbLed()* to do this, or just make it intelligent by checking for 0 or 255, but that might increase size or slow down the code.

There should also be some function or #define or something that the user can use to determine what hardware the sketch is running on, so they can know whether the RGB LED or just the single LED is available, whether the speaker has PWM capability, plus any other differences, so as to allow them to have their code behave accordingly.

yyyc514, 2016-02-08:
> There should also be some function or #define or some other "official" way that the user can use to determine what hardware the sketch is running on, so they can know whether the RGB LED or just the single LED is available, whether the speaker has PWM capability, plus any other differences, so as to allow them to have their code behave accordingly.

That's what the define does and board file does. The user will have to pick the correct hardware in the Arduino IDE menu when they compile their sketch.  Kevin has said he wants production to be the default if someone just downloads the library blind and picks Leonardo.

I think we ought to really try to encourage use of the board files though.

I'm not sure if there is a great way to detect the versions via software.  I know it would result in a slightly larger compile since you need code for both versions in the binary always.  If someone knows how to probe the difference though I'd definitely consider that.

I'm working on this tonight or tomorrow.

yyyc514, 2016-02-08:
> AFAIK it's just the RST, CS, DC, buttons, and speaker pins that changed - the rest of the hardware is unchanged,yes?

@rogosher Do you have an exact list of the changes (and pin outs) or do I need to get that from Chris?

yyyc514, 2016-02-08:
This obviously needs to happen soon, but it also doesn't have to correspond with 1.0... version numbers are just numbers on github.... it could be that we release 0.9 with this and then 1.0 a week later with other fixes.  1.0 doesn't have to co-incide with the release of production hardware.

yyyc514, 2016-02-08:
> It would be nice if there were one or more functions for controlling the RGB LED (available only if compiled for the production version).

I dunno if my beta has the RBG, I'll have to ask Kevin.  But yes that's obviously something we should add to the hardware API, no brainer.  For the DEVKIT I would suggest it average the values or use the highest value and just set the LED to that - if our single LED is PWM.  If not then set it high/low depending on 0 or > 0.

MLXXXp, 2016-02-08:
The LED on the DevKit is on PB0/SS/RXLED Arduino digital Pin 17. It isn't PWM capable.

yyyc514, 2016-02-08:
Ah, is that why it blinks all the time when plugged into USB, transmit?

MLXXXp, 2016-02-08:
>That's what the define does and board file does. The user will have to pick the correct hardware in the Arduino IDE menu when they compile their sketch.

Which *#define* is it (I haven't looked very hard for this)? Will it be standardised and documented for making decisions in sketches?
Something like:
```cpp
#ifdef ARDUBOY_HARDWARE_PRODUCTION
  // do stuff with the RGB LED
#else // Must be a DevKit
 // do the best we can with the single LED
#endif
```

MLXXXp, 2016-02-08:
>Ah, is that why it blinks all the time when plugged into USB, transmit?

Yup.

yyyc514, 2016-02-08:
> Which #define is it (I haven't looked very hard for this)? 

Currently it only defines DEVKIT... when production code is added it'll define that instead and you'll have to ifdef depending on the hardware.  Unless someone submits a software only sniffing solution... to me that becomes a LOT more important IF/when we start shipping binaries though.

yyyc514, 2016-02-09:
https://github.com/Arduboy/Arduboy/pull/63

closed 2016-02-09 by yyyc514