Ar-zz-duboy/Arduboy #70: FloatyBall or FlappyBall?

MLXXXp, 2016-02-09:


MLXXXp, 2016-02-09:
Also, we may want to get rid of the little *ARDUINO* logo that's displayed on start up, especially if there's going to be an automatic *Arduboy* logo.

MLXXXp, 2016-02-09:
Actually, I think this game should be removed unless it's tuned a bit. At the speed the ball rises and falls, I find it virtually unplayable.

rogosher, 2016-02-09:
It's not exactly a shining beacon of programming magnificence, but people do play it/ask for it.

I can fix those two issues.

yyyc514, 2016-02-13:
I put it under proper frame management (at 30 FPS as it was designed for) but man you're right it's unplayable. :-)

yyyc514, 2016-02-13:
To clean it up you'd make the bird a bitmapped sprite with animation frames and bump this to 60 FPS and then adjust the physics.  You could also have really nice bitmaps for the columns vs them being simple rects.  Lots of opportunity here once you fixed the physics.

rogosher, 2016-02-13:
This was discussed a bit with @bateske, and we can just take out flappy bird. It will probably be easily found on codebender etc in the future. Or an arduboy-extras repo.

closed 2016-02-16 by rogosher