Ar-zz-duboy/Arduboy #71: Compile warnings due to always_inline attribute

MLXXXp, 2016-02-09:
The library generates three compile warnings due to using the *always_inline* attribute.
```
core.cpp:145:6: warning: always_inline function might not be inlinable [-Wattributes]
 void ArduboyCore::safeMode()
      ^
core.cpp:112:6: warning: always_inline function might not be inlinable []
 void ArduboyCore::bootLCD()
      ^
core.cpp:93:6: warning: always_inline function might not be inlinable [-Wattributes]
 void ArduboyCore::bootPins()
      ^
```
I don't like to see compiler warnings. It make me wonder if it might be more serious than just a warning, such as what I described in pull request #69. Is there something that can be changed to eliminate these warnings?

If you feel that the code is correct and the warnings can be ignored, via a quick search I found a possible solution using pragmas .  Example ([described here](http://stackoverflow.com/questions/3378560/how-to-disable-gcc-warnings-for-a-few-lines-of-code)):
```c
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wunused-result"
    write(foo, bar, baz);
#pragma GCC diagnostic pop
```
I tried it, using *-Wattributes* and containing the entire function, and it worked.

rogosher, 2016-02-09:
I think we might look at this example: http://stackoverflow.com/questions/32432596/warning-always-inline-function-might-not-be-inlinable-wattributes

See the bottom reply.

yyyc514, 2016-02-10:
I'll look into this, but it's 'causing no harm - and it is doing what it's intended to - inline methods that are never meant to be called outside the init sequence - saving space in flash.

yyyc514, 2016-02-10:
Ref: https://github.com/Arduboy/Arduboy/pull/72

yyyc514, 2016-02-10:
If there is some problem with my PR and it's not fixable any other way then I suggest we try `#pragma GCC diagnostic ignored`.  I should point out that I think the default is for the compiler not to even show warnings though.  Mine was turned off and I had to turn it on manually at least.

yyyc514, 2016-02-10:
Closing to reopen if PR is inadequate.

closed 2016-02-10 by yyyc514

MLXXXp, 2016-02-10:
> I should point out that I think the default is for the compiler not to even show warnings though.

There's a setting in the IDE under *File -> Preferences* in the *Settings* tab:
*Show verbose output during:* (with check boxes for *compilation* and *upload*)

Setting it for compilation will show the warnings, which is how I keep it set.

rogosher, 2016-02-10:
Thanks @MLXXXp thats a good inclusion for the issue, I didn't know about that setting. Ill admit, I'm the atypical user using a makefile to build the arduboy programs when its for personal use.