Ar-zz-duboy/Arduboy #74: Look into binary compatibility for devkit and production sketches

yyyc514, 2016-02-10:
Anyone interested?  What I mean is removing the compiler defines and letting the software detect which version of the Arduboy it's running on.  This would be more important in the future when games are readily swapped in a out via some tool, or on SD.

Questions:

- How much would this increase the compiled flash size?

Ideas on how:

- Poll a floating pin that is wired differently between the two boards (floating on one, connected on another) and would give clear differences
- Check the battery voltage (2.5 vs 3.7), though this would fail when plugged into USB
- Check bootloader? (though I assume right now it's the same across all)

MLXXXp, 2016-02-10:
> Poll a floating pin that is wired differently between the two boards (floating on one, connected on another) and would give clear differences

You can't trust floating pins.  They might read always high or low. Reading a floating analog input a few times and checking for changes might work but I still wouldn't trust it 100%.

>Check the battery voltage (2.5 vs 3.7), though this would fail when plugged into USB

It would also fail for people like me, who've built clones with a different power supply (3 AAA cells in my case).

>This would be more important in the future when games are readily swapped in a out via some tool, or on SD.

It's probably best to just maintain two game pools and compile twice.

rogosher, 2016-02-10:
I think we might also look at the volume of dev kits out there in relation to the expected number of production units to be in the wild. It may not be a feature that many people end up using/asking for in a few months. Hard to say.

yyyc514, 2016-02-11:
> I think we might also look at the volume of dev kits out there in relation to the expected number of production units to be in the wild. It may not be a feature that many people end up using/asking for in a few months. Hard to say.

This is more of a research item if there is a binary compatible Arduboy 1.1, 1.2... Still on the Atmega32u, etc... if we start to build a game library of binaries they shouldn't just fail to work if you got your 99% the same Arduboy 1.2.

But again this is a longer-term research style item for someone with the time.

> They might read always high or low. 

Well they are supposed to float, but if they were wired into something they would have entirely different behavior... so you can probably already use the difference in button wiring to detect a devkit or release... but the real question becomes how to play for future hardware that might change the pins around again that we don't know about yet.

Maybe it's not doable - i.e. the new hardware would have to purposely be backwards compatible.  Some of this could be pushed into settings in an area of the bootloader but no one has seemed super interested in hacking on the boot-loader yet (or not knowledgable enough).

For example you could imagine a setting in the bootloader telling you the ports, pins, hardware config in 16-32 bytes... then any compatible software could be loaded, check that config, and then setup the hardware properly...

Even potentially entirely new LCD screen, just store the LCD boot program in the bootloader, with a flag for whether it's SPI or i2c, etc...  SO maybe I should have posted this on the forums. :-)

closed 2016-05-27 by rogosher

yyyc514, 2016-07-02:
This was a research item, why was it just closed?