Ar-zz-duboy/Arduboy #76: Control of TXLED and RXLED

MLXXXp, 2016-02-10:
It looks like the production Arduboy has both a TXLED and RXLED wired on the same pins as the Arduino Leonardo, Micro and Pro Micro (and likely others). Therefore, I assume the Arduboy's  *pins_arduino.h* file will include the definitions for controlling these LEDs as others do. E.g.
```cpp
#define TX_RX_LED_INIT	DDRD |= (1<<5), DDRB |= (1<<0)
#define TXLED0			PORTD |= (1<<5)
#define TXLED1			PORTD &= ~(1<<5)
#define RXLED0			PORTB |= (1<<0)
#define RXLED1			PORTB &= ~(1<<0)
```
```cpp
#define LED_BUILTIN_RX 17
#define LED_BUILTIN_TX 30
```
For reference: [Example 1: Blinkies!](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide/example-1-blinkies)

Is this adequate to allow a sketch to control these LEDs or should we include library functions to make it easier?

Note that the RXLED is on the same pin as the DevKit's LED.

MLXXXp, 2016-02-10:
>For reference: [Example 1: Blinkies!](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide/example-1-blinkies)

Note that, contrary to that article, you can control TXLED as digital Pin 30 (like RXLED is Pin 17). I think this was added in Arduino version 1.5+

yyyc514, 2016-02-11:
What file are these in?  I didn't see them when I looked.  I think eventually we should have a nice API wrapper so we can hard code the PORTS and bit stuff rather than people using digitialWrite (which is super expensive).

I was planning on getting to it but hadn't decided an API yet... one method, two?  Boolean?  They aren't hardware PWMable.

MLXXXp, 2016-02-11:
>What file are these in?

The file is *pins_arduino.h* which is located in a subdirectory of the *variants* directory for the board that you are using. The *variants* directory is somewhere under a *hardware* directory. The location of *hardware* differs depending on whether it's an official Arduino installed as part of the IDE or a third party board installed as an add-on. The locations are probably also dependent on the host O/S.

yyyc514, 2016-02-11:
I'm very familiar with that file... it's a trusted companion... but I must have missed the TX/RX stuff... looking again now.

yyyc514, 2016-02-11:
```
    void setRGBled(uint8_t red, uint8_t green, uint8_t blue); // already coded
    void setRXled(bool onoff);
    void setTXled(bool onoff);
```

And we just wrap those defines nicely with an API.  Thoughts?

MLXXXp, 2016-02-11:
The [*Arduino Style Guide for Writing Libraries*](https://www.arduino.cc/en/Reference/APIStyleGuide) states:
>Use read() to read inputs, and write() to write to outputs, e.g. digitalRead(), analogWrite(), etc.

Which would mean using *write* instead of *set*.
```cpp
void writeRGBled(uint8_t red, uint8_t green, uint8_t blue); // already coded
void writeRXled(bool onoff);
void writeTXled(bool onoff);
```
Or the more compliant postfix instead of prefix: *RGBledWrite()*, etc.?

And should we provide complimentary *read* functions?

And if we decide on this, then maybe *readButtonState()* or *readButtonStatus()*  (or the postfix versions) instead of the *pollButtonState()* or *pollButtonStatus()* that you are proposing in place of *getInput()*.

Strangely, the guide says later:
>**Try to avoid boolean arguments.** Instead, consider providing two different functions with names the describe the differences between them.

But maybe that refers to functions with multiple parameters?? Or non-I/O related functions??

yyyc514, 2016-02-11:
> Or the more compliant postfix instead of prefix: RGBledWrite(), etc.?

I don't see where it mandates the order. And pins are inputs and outputs, lights are lights.  They can be wired to pins yes, but those low level functions are for working with pins - and the concepts are named accordingly.  We don't write to the arduboy screen, we draw to it... higher level gets to name things higher level. :-)

> And should we provide complimentary read functions?

I don't think that's necessary, if someone wants to do crazy animation or a tight loop that should probably be tracking the status themselves in a counter, not polling for it each time.  Plus RGB isn't an input that you can just poll... I'm pretty sure you could get those numbers back by reading the IO counter ports and inverting, but it's not analogous to analogRead() or anything - it's a whole other thing.

I think you could argue the boolean thing both ways. If you're doing that though you're not really "writing" anything anymore (even at the function level)... write(true) is maybe writing something... TXon() and TXoff() is just state toggling... I wouldn't call it `writeTXon` or `TXonWrite`, that's just silly.

So what about:

```
  ledTXon(); 
  ledRXoff();
  // ...
  
```

yyyc514, 2016-02-11:
And lets remember it's a guide, not a Bible. :)

yyyc514, 2016-02-11:
```
void writeRGBled(
```

While you might write the red inverse duty value, write the green inverse duty value, and write the blue inverse duty value when you do them all together now you're doing a new thing.  You're setting a color, a much higher level concept. :-)

I also like `setRGBcolor(r,g,b)`.

MLXXXp, 2016-02-11:
>And pins are inputs and outputs, lights are lights.

So what type of things do you think the guide is referring to when it says:
>Use read() to read inputs, and write() to write to outputs

If it's not for creating more specific functions for things attached to pins, what are they referring to?

yyyc514, 2016-02-11:
> If it's not for creating more specific functions for things attached to pins, what are they referring to?

It's for writing to pins.  They couldn't name it any differently because they don't KNOW what you are attaching.  If you're attaching a printer you might want to print... if you're attaching a gun you might want to shoot... the higher level the function the higher level the concept.  Your api would be print("Bob") or shoot(5)... look at some of the other Arduino libraries, not just the guide.

if you're selling someone a board with 20 switches they might all say "switch one", "switch two", but someone is going to write "fridge", "washer", "drier" on the switches, and they'd be insane not to. :)

yyyc514, 2016-02-11:
Look at the mouse api for example.  You don't write mouse status to USB, you `click` and `move` and `press` and `release` things.  The functions match the function.

I think you missed the very top of the guide:

```
Be kind to the end user. Assume you are writing an API for an intelligent person who has not programmed before. Come up with a clear mental model of the concept you’re working with, and the terms and functions you will use.
```

There are things you write and read, and then they are things you shoot, click, launch, boot, process, etc... and sometimes if you're at a higher level you are setting things, not writing them... search the entire Arduino source for set.*( and you'll find plenty of set, including setTX and setRX for one example, for setting those pins.

MLXXXp, 2016-02-11:
*writeRGBled()* or *writeRGBcolor()* read as clearly to me as *setRGBled()* or *setRGBcolor*. (Except, being Canadian, I would keep entering *colour* and have to go back and correct it to *color* :wink:)

Actually, thinking further, *writeRGBcolor()* (or *set*) is a bit vague. RGB alone doesn't really describe a specific object. If you're going to include *color* in the function name, it should also indicate that it's an LED, such as *setRGBledColor()*. So I'd just use *setRGBled()* without *color*. Besides, we're actually setting intensity as well as color. (We might even want to overload the function, at a later date, to allow [HSV or HSL](https://en.wikipedia.org/wiki/List_of_color_spaces_and_their_uses#Hue_and_saturation), or something else.)

From a user standpoint, I don't look at it as being three separate LEDs in one package. I see it as a single point of light that can be set to many colors and intensities, like a pixel on a display. So just like I *write* to a RAM location, I'm writing to a pixel or other light source.

Anyway, I personally don't care whether you use *set* or *write*, or prefix or postfix, or separate functions instead of boolean arguments, I was just pointing out what it says in the guide (at least, my interpretation). To me, *set* and *write* are synonymous, as are *get* and *read*. I've had no problem understanding any of them and all would be fine by me.

MLXXXp, 2016-02-11:
>It's for writing to pins. They couldn't name it any differently because they don't KNOW what you are attaching.

The could have named them *setDigitalPin()*, getDigitalPinState(), *setAnalogOutput()*, *getAnalogInput()*. :wink:

OK, I'll shut up now :smile:

yyyc514, 2016-02-11:
As I said they use set* in plenty of other places... it's all contextual.

yyyc514, 2016-02-11:
@rogosher Any thoughts on:

```
 ledTXon(); 
 ledRXoff();
```

MLXXXp, 2016-02-11:
>Any thoughts on:
 ledTXon(); 
 ledRXoff();

If the *TXLED0*, *TXLED1*, *RXLED0*, *RXLED1* macros are kept in the the Arduboy version of *pins_arduino.h*, they will do same thing. @yyyc514, I know that you don't like this kind of obfuscation, but they're already documented and accepted by the Arduino community, so I don't have a problem with just using them directly.

Then again, adding the functions doesn't hurt, and they're a bit clearer than the macros. They won't add to code size if they're not used.

Personally I would prefer single functions (regardless of the actual name) with a boolean on/off parameter, regardless of if you interpret the style guide as not recommending them.

```cpp
void setRXled(boolean on);
void setTXled(boolean on);
```
They can be used to force the LED to a desired state as easily as two separate functions, but it makes it a bit easier to restore a previously saved state:
```cpp
void setRXled(savedLEDstate);
```
instead of
```cpp
savedLEDstate ? ledRXon() : ledRXoff();

// or more likely for a programmer unfamiliar with using the conditional expression:
if (savedLEDstate == ON) {
  ledRXon();
} else {
  ledRXoff();
}
```

yyyc514, 2016-02-11:
I'll think about it more. One advantage of functions (other than being easy to find in the Api docs) is that if you use them a few times they start to save you flash space. Those defines just get added to your code over and over again. 

Not sure why arduino itself doesn't provide a helper for this. Maybe it assumes using the LEDs for serious stuff someone would know all this and act accordingly?

MLXXXp, 2016-02-11:
>Not sure why arduino itself doesn't provide a helper for this.

Because the TX and RX LEDs were put there mainly to provide USB communication status. Their use for other purposes isn't common, and possibly not even encouraged.

yyyc514, 2016-02-11:
Except we are really custom hardware - and they have no purpose at all when not connected to USB. But I guess I see why they didn't encourage going crazy with them. I could see going either way on this.

yyyc514, 2016-02-12:
OK, I'm in in the air now - does anyone else have an opinion on whether the TX/RX should be included?

MLXXXp, 2016-02-12:
It doesn't matter much to me if they're included or not, but as I've already said, if you **do** add them, I'd prefer two functions that accept a boolean argument instead of four separate *on* only or *off* only functions

yyyc514, 2016-02-12:
I think I'm going to side with Arduino for now.  These are hardware LEDs for TX/RX and if someone wants to use them it's also easy enough to use the provided defines.

closed 2016-02-12 by yyyc514