Ar-zz-duboy/Arduboy #78: rename getInput() to pollButtons()

yyyc514, 2016-02-11:
Or pollButtonState()?  pollButtonStatus()? getInput is too generic and makes me write library code like:

```
uint8_t button_state = getInput();
if (button_state & LEFT_BUTTON)
```

I do that because to me getInput is opaque and I want clarity.  What input exactly am I getting?  But then I realized the real problem is the name of the method, it could be more descriptive.

Not many should use this directly but we should still have the best name possible

yyyc514, 2016-02-11:
```
readButtonState() // ?
readButtons() // ?
```

I could imagine read used here.

MLXXXp, 2016-02-11:
Like I said in PR #23, I'm kind of like just `buttonState()`.

yyyc514, 2016-02-12:
The word state is very programmery, but so is poll.  Condition is silly though.  Is there  much better word I'm not thinking of?

MLXXXp, 2016-02-12:
>The word state is very programmery

I don't know if I agree about that. *State* has pretty much the same meaning outside of programming. E.g. "*State of the Union*", "*In a confused state*". So, I think *buttonState* is fairly descriptive. Even the Doxygen comments for *getInput* currently begin with:
`// get current state of buttons (bitmask)`

And @yyyc514, you yourself used `button_state = getInput()` when you wrote function *pressed()*.

Besides, people using the function **will** be programmers, or if they're "programmers in training" they should be expected to become familiar with the meaning of *state* in this context.

I agree that *condition* doesn't sound right. The only other I can think of  is *buttonPositions()*, which I'm not too keen on.

yyyc514, 2016-02-13:
`buttonsState()`  I like.  It's clear that it's polling multiple buttons (not just returning one) and again this is low-level and people using it should understand what it does.

MLXXXp, 2016-02-13:
>`buttonsState()` I like

That's fine with me then. :smile:

closed 2016-02-13 by yyyc514