Ar-zz-duboy/Arduboy #90: Shipped game included in examples

MLXXXp, 2016-02-16:
What game is the production Arduboy shipping with?

I think it would be a good idea to have this game included in the library examples. After a user loads something else, they may want go back to the default game. Having it as an example makes it easy to find and re-load.

rogosher, 2016-02-16:
That is a good question. Part of me thinks it might be one of JO3RI's.

MLXXXp, 2016-02-16:
Even if it is JO3RI's, if it's the default game shipped with the Arduboy, then it should be maintained as part of the Arduboy source.

rogosher, 2016-02-16:
I agree.

JO3RI, 2016-02-16:
https://github.com/TEAMarg/ID-28-Blob-Attack

yyyc514, 2016-02-17:
> then it should be maintained as part of the Arduboy source.

I don't see where that is a requirement if it's already on Github and separate maintained - link to it - that's what the web is for.

rogosher, 2016-02-17:
This is beyond the scope of the Arduboy library.

closed 2016-02-17 by rogosher