Ar-zz-duboy/Arduboy #94: ab_logo.h shouldn't  be included in sketches

MLXXXp, 2016-02-18:
When a developer uses *Sketch -> Include Library -> Arduboy* from the IDE menus, `#include <ab_logo.h>` is added to the sketch (along with `#include <Arduboy.h>`). This header shouldn't be included in a sketch.

The reason *ab_logo* is included is because the IDE includes all *.h* file residing directly in the *src* directory.
Possible ways to fix this:

- Move *ab_logo.h* to a subdirectory under *src/* (such as *bitmaps*) and in *Arduboy.cpp* change `#include "ab_logo.h"` to `#include "bitmaps/ab_logo.h"`

- Change *ab_logo.h* to *ab_logo.c*, as *glcdfont.c* is. Why is one a *.h* file and one a *.c* file anyway? They both do the same thing, so should both have the same extension. The library should be consistent.

Note that if we decide that *glcdfont.c* should be *glcdfont.h*, it should be moved to a subdirectory under *src/* as well, for the same reason.

rogosher, 2016-02-18:
I do like moving from a header tile to a source file, as I don't see this file as being something many people will change the source for, and it's a small bit of source.

MLXXXp, 2016-02-18:
So should *glcdfont.c* be changed to *glcdfont.h*?

Is *bitmaps* a good name for the directory to put both files in?

yyyc514, 2016-02-18:
One difference is C files are going to be auto compiled and .h files are not, but since both of these files only add a variable that is either referenced or unreferenced I don't think it should really matter to the compiler... so I think them both being .c wouldn't necessarily be harmful... but I really don't know the "proper" C opinion on this.  It seems in most of the source I find/read that large amounts of data/images/etc are put in the headers.

I think we should find out and then either rename both, or make both .h.

I preset `assets` or `resources`.  Bitmaps is too specific.

yyyc514, 2016-02-18:
If we're still getting carried away with dirs maybe put the font in `fonts`. :-)

yyyc514, 2016-02-18:
I'd love to see us ship a non ugly font, but no time to work on it.

MLXXXp, 2016-02-18:
>I don't think it should really matter to the compiler... so I think them both being .c wouldn't necessarily be harmful... but I really don't know the "proper" C opinion on this.

It's obviously not harmful, since it works for *glcdfont.c*.

My personal rule is to never put anything in a header file that actually generates code or sets values in storage. A *.h* file should be used to define prototypes or templates for things, not actually create things. I would have used a *.c* or *.cpp* file. I found a few references on the Web that agree with this. I didn't see any that said a *.h* file should be used.

Changing *ab_logo.h* to *ab_logo.c* would mean we wouldn't have to move it to a subdirectory.

rogosher, 2016-02-18:
I agree with the notion of a header being used explicitly for definitions. C++ really doesn't ever want you to put source in a header file; a project's source should always be in a source file. That's pretty [cut and dry in the C++ world](http://stackoverflow.com/questions/583255/c-code-in-header-files).

So go towards `.c` files for the logo and font source.

I do like that idea of a less specific folder, `assets` or `resources` do seem to fit better than bitmap. If a goal is to reduce the folder count going forward then a more abstract folder name could catch `resources`/`assets` in the future. But these nouns do suggest some sort of binary like an image. `helpers`? Not sure in this moment.

MLXXXp, 2016-02-18:
>I do like that idea of a less specific folder, assets or resources do seem to fit better than bitmap.

Like I've said, if we are going to change *ab_logo.h* to *ab_logo.c*, then *ab_logo.c* and *glcdfont.c* can both remain directly in the *src* directory without any issues. Therefore, there's no need to decide on a name or create a new subdirectory at this time.

rogosher, 2016-02-18:
Cool beans, time for my nap ; )

MLXXXp, 2016-02-19:
PR #98 created.

closed 2016-02-19 by yyyc514