Ar-zz-duboy/Arduboy #95: The assets directory should be, or be under, extras

MLXXXp, 2016-02-18:
In keeping with the [*Arduino IDE 1.5 Library Specification*](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#extra-documentation), I think that files *arduboy_logo.png* and *arduboy_screen.png* should be moved into a directory named *extras*, and the *assets* directory removed. The specification doesn't mention allowing directories in the root other than those described, so it's possible that it could cause problems in the future.

If we feel that it's desirable to have a directory specifically named *assets* for the above files, it could be made a subdirectory in the *extras* directory.

Quoting the specification:
>Arduino libraries will be composed of a number of folders. Each folder has a specific purpose (sources, examples, documentation, etc). Folders not covered in this specification may be added as needed to future revisions.

and:
>An extras folder can be used by the developer to put documentation or other items to be bundled with the library.

rogosher, 2016-02-18:
Sounds good to me. If I make a change soon, I will include this.

MLXXXp, 2016-02-18:
>Sounds good to me.

What sounds good?
`extras/<files>`
or
`extras/assets/<files>`

yyyc514, 2016-02-18:
Please don't move this.  My utility looks for it in assets.  Is it really harming anything?

yyyc514, 2016-02-18:
The spec is kind of quiet on if you can have your own directories - it doesn't explicitly say no.  If we're going to move it it should be `extras/assets`.

MLXXXp, 2016-02-18:
>The spec is kind of quiet on if you can have your own directories - it doesn't explicitly say no.

True, but the spec does say "*Folders not covered in this specification may be added as needed to future revisions.*". There's an outside chance, however slight, that someday they'll add a folder named *assets*, which causes us problems due to us already having an *assets* folder containing files that the revised spec doesn't allow to be there.

On the other hand, the spec guarantees that we can put anything we want under *extras* without the chance of causing future issues.

I agree that putting the files in `extras/assets` is better than just putting them directly in `extras`. It helps to categorise their purpose.

rogosher, 2016-02-18:
It's probably better to err on the side of conservative when it comes to matching and meeting specs. It will help the project become a good example for anyone else wanting to do similar projects. There are also a lot of young eyes on this project, so ya know, it pays to do it by the book here.

Also, `extras/assets` looks reasonable to myself.

MLXXXp, 2016-02-18:
PR #97 created.

closed 2016-02-19 by rogosher