Ar-zz-duboy/Arduboy #103: Setup a mailing list or separate forum for Arduboy developers

yyyc514, 2016-02-21:
Make it easier to notify of new library changes or breaking changes, etc.  I think we ought to encourage authors to compiled against the core lib and follow it when it upgrades (to make sure their stuf still compiles, etc).

At least until we have great binary installers.

MLXXXp, 2016-02-21:
I'm all for this. A separate forum on arduboy.com may be best, but if not, [Google Groups](https://groups.google.com/forum/#!overview) works well for this sort of thing.

>At least until we have great binary installers.

Even **when** we have a binary installer, we should encourage authors to keep their sketches compatible with the latest API and recommended practices. Other people, especially beginners, may wish to look at and follow their code as an example.

rogosher, 2016-02-21:
You and Dreamer were added to a Developers group on the community page, I started passing out badges and the last step is to open up a private category for discussion... ... ... and now one is open. If you do not see a private 'developers' category, please let me  know via the community page. It's private right now until we decide how it should be handled. I'll also see about how to install moderators for just that category perhaps, but and 'regular' user on the forum should be able to edit or move discussions.

closed 2016-02-21 by rogosher