Ar-zz-duboy/Arduboy #105: Simple and secure binary installer for all platforms

yyyc514, 2016-02-21:
Looking at a batch file uploaded to the forums as a template:

```
@echo off
SET VAL1="C:\Program Files\Arduino\hardware\tools\avr\bin"
SET VAL2=COM5
SET VAL3="C:\Program Files\Arduino\hardware\tools\avr\etc\avrdude.conf"
SET VAL4=COM4

set PATH=%VAL1%;%PATH%

tool\reset %VAL2%
avrdude -C%VAL3% -v -patmega32u4 -cavr109 -P%VAL4% -b57600 -D -Uflash:w:test.hex:i

pause
```

I see a few questions here:

1. What software to use for the steps (including reset)... Ruby, Python, C#, Bash?  I know there are many good choices on Mac/Linux but not sure about the cross-over to Windows.
2. How to do the USB reset?
3. How to detect the proper USB port for avrdude (or maybe avrdude can figure it out if it's not passed)
4. How to detect the binary and config location of avrdude

I assume for simplicity we'd require installing the Arduino IDE, so perhaps the answer to #4 is that we'd use the default locations for the various platforms.

So how about 1, 2, and 3?

The idea being someone would add "commando.hex" to their github and when somone downloaded the source they'd have not only the source + the binary and they'd know to use the standardize upload process to upload the binary (even if they have no idea how to use the source).

yyyc514, 2016-02-21:
All of this would be pretty simple with Ruby but I'm not sure what the whole Ruby/gems situations is like on different distributions of linux... though honestly I wonder how many users on each platform we have.  I bet Mac and Windows are the most popular platforms.

yyyc514, 2016-02-21:
Google's Go is cross-platform and can produce distributable binaries...

MLXXXp, 2016-02-21:
>What software to use for the steps (including reset)... Ruby, Python, C#, Bash? I know there are many good choices on Mac/Linux but not sure about the cross-over to Windows.

Well, since the Arduino IDE is written in Java, perhaps the binary installer should follow suit.
[*Arduino and Java*](http://playground.arduino.cc/Interfacing/Java)

yyyc514, 2016-02-21:
That would suck since I refuse to install  system-wide Java since the latest huge security issue a while back - and Mac thankfully doesn't install it by default.  (yes, I'm aware the IDE uses an embedded Java)

I don't think you can just piggy-back the Java included in the IDE either... my choice would be something with minimal (or no) dependencies - and I think Java is a HUGE dependency.  We certainly don't want to bundle it (and deal with the security issues of doing that)... and I wouldn't want to force Mac users to install it either when they don't need it for anything else and when it's introduced HUGE security issues in the recent past.

yyyc514, 2016-02-21:
I mentioned Go because it can compile down to static binaries with 0 dependencies of their own.  So the user would only need to install the binary, nothing else.

yyyc514, 2016-02-21:
I would just throw something together in Ruby but not interested in doing something that won't benefit most users - and it's not a problem I have since I know how to pretty easily compile from source (and that's my preference just to look at the code anyways).

yyyc514, 2016-02-21:
Though honestly how helpful is a command line installer for most users?  I guess it's way better than building from source?

yyyc514, 2016-02-21:
https://github.com/noopkat/avrgirl-arduino

Wow, this is really close, but I dunno about packaging a nodejs app for easy use on Windows without requiring users to install nodejs, etc... I assume most linuxes would have nodejs in package manager and on Mac you can easily get via homebrew.

yyyc514, 2016-02-21:
What I'm imagining:

```
arduboy_install flashlight.hex
```

rogosher, 2016-02-21:
Ruby is gaining more traction in the IT world. Nodejs is getting more traction on the Windows machine. It's a strange world. The avrgirl stuff is pretty cool, they posted a couple things in the community section and are interested in helping as well.

Getting nodejs is pretty simple on linux, it might be hard matching versions. NPM can throw users for a loop. All of this works on windows too. 

Also, it would be a huge package, but people like puppet include ruby binary with their project. There is also Docker and Vagrant which we might use to wireframe and environment for a user. Although these techs aren't in use by an 'average' user, it's easy to imagine VMs picking up steam as a container for instruction, especially Docker.

MLXXXp, 2016-02-21:
>https://github.com/noopkat/avrgirl-arduino

To get to the point where I can run command `avrgirl-arduino test-pilot` and get the *AVRGirl Arduino test pilot* page to come up in my browser, under Ubuntu Linux 15.04, I had to:

- Using the *Software Center* (package manager) install:
  - nodejs
  - nodejs-legacy
  - npm

- install as root using command `sudo npm install -g avrgirl-arduino@latest`
not just `npm install -g avrgirl-arduino@latest`

MLXXXp, 2016-02-21:
When I run the *AVRGirl Arduino test pilot* using my home made system, which uses an Arduino Micro, I get:
`Errors: "no Arduino 'micro' found."`

For a SparkFun Pro Micro, I get:
`Errors: "no Arduino 'sf-pro-micro' found."`

yyyc514, 2016-02-21:
Did you ask it to list the USB port and then pass it a specific port?

-g is going to require sudo as it's GLOBAL install.

MLXXXp, 2016-02-21:
>Did you ask it to list the USB port and then pass it a specific port?

Yes, I listed the ports and found my device on /dev/ttyUSB0. I then tried
`avrgirl-arduino test-pilot -p /dev/ttyUSB0`

Nothing changed. I still got
`Errors: "no Arduino 'micro' found."`

yyyc514, 2016-02-22:
How does it know it's a micro?  I need to get the USB port adapter for my MacBook then I'll give it a try myself with real hardware. Should have it early next week. 

Bummer if this doesn't work though - looked so promising.

yyyc514, 2016-02-22:
http://russemotto.com/xloader/

Windows only?  Should we spend some time looking for things out there that already do this.  Why does Arduino IDE provide a hex export if it doesn't provide a "write hex" feature?

MLXXXp, 2016-02-22:
>How does it know it's a micro?

I'm guessing it's just whatever you chose from the "*Which board is it?*" pulldown list. Although, since you don't have to specify a USB port, perhaps it scans the USB ports somehow, to find out which port the device you specified is on.

When I plug in my Arduino Micro based home made Arduboy, and use
`avrgirl-arduino list`

the list includes:

```
  { comName: '/dev/ttyACM0',
    manufacturer: 'Arduino_LLC',
    serialNumber: 'Arduino_LLC_Arduino_Micro',
    pnpId: 'pci-0000:00:1d.0-usb-0:1.1:1.0',
    vendorId: '0x0x8086',
    productId: '0x0x1c26',
    _standardPid: '0x0x1c26' },
```

yyyc514, 2016-02-25:
From the forums: 

http://paulkaplan.me/HexUploader/

I gotta try this.

yyyc514, 2016-02-25:
Looks workable with only a few issues.  I reached out to the author to see if he's willing to help.  Anyone who was a Mac developer could probably also step in here and have this working in a few minutes.

AFAICT the path issue is documented in issues and we'd need to add Arduboy/Leonardo to the devices.json file.   This is just a nice little wrapper for avrdude, which is all that's needed honestly.  Install the IDE, install HexLoader, done... of course this is just for Mac - but I'd rather see one platform have a great tool than 0.

It might end up we have different tools on different platforms - the important thing is that it's easy for the end users, not that we have one super tool that spans 3 platforms.

yyyc514, 2016-02-25:
Can someone test on Windows/Linux:

https://chrome.google.com/webstore/detail/avrchick/kpbgbcocfgjbmnpplcjlcammjdkgogba?hl=en

rogosher, 2016-05-27:
Falling a little outside the scope of this repository. Anyone interested in pursuing this immediately should see @noopkat's repository https://github.com/noopkat/arduboy-electron.

closed 2016-05-27 by rogosher

noopkat, 2016-05-27:
@MLXXXp hello 👋 

I know this has just been closed, but I didn't know about this thread beforehand until I was mentioned.

Thanks for giving avrgirl arduino a try. The reason that test pilot was not finding the board was because of the version of the serialport dependency I was using for the library was outputting double `0x` in the vendor and product id's (as you can see in your list output, very handy thanks for pasting this in). So when avrgirl arduino scans for the right device by product id, it skips out as it's not exactly what it was expecting. Perhaps I can look into whether it's robust enough to just skip the `0x` instances, but I believe this bug in the serialport library is already fixed 😁 

I would love to work more with y'all on this kinda of stuff for the arduboy, so feel free to @ mention me if you ever have questions, should you decide to give it a try again.

Thanks! 😄