Ar-zz-duboy/Arduboy #108: Speaker needs both pins set to OUTPUT

atomic-salamander, 2016-02-23:
Both speaker pins must be set to OUTPUT, or neither will work, like so:
```
pinMode(PIN_SPEAKER_1, OUTPUT);
pinMode(PIN_SPEAKER_2, OUTPUT);
```
This is already done in the Arduboy audio library but if someone wants to make their own and doesn't plan on using both pins it can be very difficult to track down why the speaker isn't working.

MLXXXp, 2016-02-23:
This is only true for the production Arduboy. The DevKit has the two pins tied to one speaker terminal, with the other terminal tied to ground. If you set both pins to outputs on the DevKit, the processor could be damaged.

To address the problem of someone using a different audio library, it would probably be a good idea to initialize the speaker pins in *boot()*. The production Arduboy should have the pins set to ouputs, and the DevKit to inputs (they should default to inputs, but it can't hurt to set them again, to make sure). I've submitted PR #109 to do this.

The *ArduboyAudio* class should probably be modified to only allow one channel for the Devkit.

yyyc514, 2016-02-24:
@atomic-salamander You're sketch should still be calling Arduboy.audio.begin() so that the EEPROM registered audio/volume settings can be honored.  If you call audio.begin() then the pins should be setup for you.

yyyc514, 2016-02-24:
This will be a bit more obvious after https://github.com/Arduboy/Arduboy/pull/106 but audio is already separate from tunes. (other than currently sharing the same files)

yyyc514, 2016-02-29:
Are there any power implications to have the pins OUTPUT vs INPUT?

MLXXXp, 2016-02-29:
Setting the speaker pins both to outputs (on the production Arduboy) could possibly draw less power than with either of  them being an input. The reason is that the piezo element can work as a microphone as well as a speaker. If the "speaker" picks up noise it could generate AC voltages on the pins. If this voltage falls somewhere between a high and a low on an input, this input could draw current internally.

Even if the speaker doesn't generate an AC voltage, it appears as a DC high impedance, so the pins are essentially unconnected. The ATmega32U4 datasheet has this to say about unconnected inputs:

>**10.2.6 Unconnected Pins**
If some pins are unused, it is recommended to ensure that these pins have a defined level. Even though most of the digital inputs are disabled in the deep sleep modes as described above, floating inputs should be avoided to reduce current consumption in all other modes where the digital inputs are enabled (Reset, Active mode, and Idle mode).

The high impedance of the speaker (along with the capacitor in series with it) means the pins are essentially unconnected/floating. Setting them to outputs will put them at a "defined level".

Also, having them always initialised to outputs, (as I recommended we do in the *pinBootProgram[]* array), will help guarantee that sounds will be produced when functions or libraries are used that aren't aware that the speaker is connected across two pins, such as the Arduino built-in *tone()* function. The user won't have to set the unused speaker pin to an output themselves. We'll get less "why don't I get any sound when I use *tone()* or the generic *xxx* sound library written for Arduinos?".

**However**, on the DevKit, because the two speaker pins are shorted together, it is unsafe to set them both to outputs. When one is set as an output, the other should always be set as an input. If one is set to output *high* while the other is output *low*, the full maximum current that an output can provide will flow between the pins, possibly causing overheating and damage. They are safer to be initialised as inputs (or left alone if it's known that they default to inputs).

yyyc514, 2016-02-29:
The current released libs (and going back some time) for devkit sets both pins open and plays multi-channel music on them.. so that means high/low would be quite common.  Seems no harm so far?

Lucky?

MLXXXp, 2016-02-29:
It just means the chip is quite robust. Has anyone measured the supply current when this is happening? I bet it jumps quite a bit. Not good for the battery. It probably also results in a lower volume than if the unused pin was an input.

Don't ask me to do any experiments or measurements of this. I don't want to risk damaging my hardware. (On my home made system, I only connect the speaker to one pin at a time.)

yyyc514, 2016-02-29:
The electrical stuff isn't my speciality... can someone else confirm this and if so then we should only init a single port (by default) on the DevKit - which also means no two tone music.

yyyc514, 2016-02-29:
@bateske Does this sound right regarding the speaker and devkit?

MLXXXp, 2016-03-01:
Don't take it only from me then.

[Digital Pins](https://www.arduino.cc/en/Tutorial/DigitalPins). See the second paragraph in the *Properties of Pins Configured as OUTPUT* section.

[10 Ways to Destroy An Arduino](http://www.rugged-circuits.com/10-ways-to-destroy-an-arduino/). See Method 2.

>which also means no two tone music

That was always the case for safe operation. The DevKit also can't do PWM audio. It's limited compared to the production Arduboy. The speaker was put across two pins and moved to PWM capable pins for a reason.

yyyc514, 2016-03-01:
> However, on the DevKit, because the two speaker pins are shorted together, it is unsafe to set them both to outputs.

How is this different in production again?  Aren't they both shorted together right in front of the speaker?  I don't know what TP26 and 27 are on the schematic.

MLXXXp, 2016-03-01:
On the production Arduboy, one lead of the speaker goes to Pin 13 and the other lead of the speaker goes to Pin 5. (Actually, a capacitor, which might have been changed to a resistor, has been added between Pin 5 and the speaker lead but that won't really affect much; just the volume or the tone.) TP26 and TP27 are the points on the circuit board that each of the speaker leads are attached to. I'm not sure what SJ1 (a solder jumper) is for, or where it goes to, but I assume it would be left un-jumpered, so wouldn't affect anything.

On the DevKit, Pins A2 and A3 are shorted together and both connected to one of the speaker leads. The other speaker lead is connected to ground.

yyyc514, 2016-03-01:
So if you do inverse pin flipping to make the sound louder, how is that working?  Aren't you sinking voltage into whichever pin is low?  Again, sorry, not an electronics guy.

Sounds like we need to officially declare one of the Devkit pins for use and the other not for use.  Thoughts?

MLXXXp, 2016-03-01:
>So if you do inverse pin flipping to make the sound louder, how is that working?

It can sometimes be a bit difficult to comprehend, but let's see if I can explain it.

Sound is generated by the speaker when the voltage **changes** between the leads of the speaker, at the frequency of the sound produced. The magnitude of this voltage "swing" determines the volume of the sound.

Let's talk about the swing in terms of how speaker lead 1 sees the change in voltage between it and speaker lead 2. The formula for the voltage on lead 2 with respect to lead 1 at any given moment in time is: voltage on lead 2 minus the voltage on lead 1 (V2 - V1). We'll assume that an output LOW is 0V and an output HIGH is 5V. Speaker lead 1 is on Pin 1 and lead 2 is on Pin 2

For the case where Pin 1 remains at a fixed LOW and we only toggle Pin 2:

- At first, both pins are LOW, so the voltage that lead 1 sees on lead 2 is 0V - 0V = 0V.
- We then make Pin 2 HIGH (while Pin 1 remains LOW) so the voltage changes to 5V - 0V = 5V
- So, the voltage swing on lead 2 with respect to lead 1 is 5V - 0V = 5V

Now, for the case where we're flipping the pins, so one is always HIGH when the other is LOW:

- At first, Pin 2 is HIGH and Pin 1 is LOW, so the voltage that lead 1 sees on lead 2 is 5V - 0V = 5V.
- We then make Pin 2 LOW and Pin 1 HIGH, so the voltage changes to 0V - 5V = -5V
- So, the voltage swing on lead 2 with respect to lead 1 is 5V - (-5V) = 10V
- This is twice the voltage swing that we got by only toggling one pin, thus resulting in a higher volume.

MLXXXp, 2016-03-01:
>Sounds like we need to officially declare one of the Devkit pins for use and the other not for use.

Absolutely, as it always should have been.

For the DevKit, we should initialise (or leave) Arduino Pin A3 as an input, initialise *PIN_SPEAKER_1* as an output, and remove the following *#define*s for the DevKit so sketches will get an error if they try to use them:
```cpp
#define PIN_SPEAKER_2 A3
#define PIN_SPEAKER_2_PORT &PORTF
#define PIN_SPEAKER_2_BITMASK _BV(4)
```

We should also caution developers that Arduino Pin A3 should never be set as an output (even for the production Arduboy, since A3 is the *DOWN* button. 

Kevin B., or whoever designed the DevKit, should never have shorted two pins together in the first place, and whoever wrote the audio initialisation code in the library should never have set both pins to outputs.

yyyc514, 2016-03-02:
> We should also caution developers that Arduino Pin A3 should never be set as an output (even for the production Arduboy, since A3 is the DOWN button.

This shouldn't happen unless someone is compiling a sketch wrong, and then they are likely not going it intentionally - so I'm not sure where you'd put such a warning?

yyyc514, 2016-03-02:
PR: https://github.com/Arduboy/Arduboy/pull/124

closed 2016-03-02 by yyyc514

MLXXXp, 2016-03-02:
>so I'm not sure where you'd put such a warning?

There's already a warning in the *Piezo Speaker* section of the [Dev Kit Hardware](http://community.arduboy.com/t/dev-kit-hardware/174) topic on the forum, so I guess that's already a good place to have it (although IMHO the warning isn't stern enough.