Ar-zz-duboy/Arduboy #117: Do development on a branch?

MLXXXp, 2016-02-29:


yyyc514, 2016-02-29:
Master is the development branch, releases are tagged and easily accessible for those wishing to use a release.  Most users should get the library from the IDE, if they are using the repository directly then they're game for pre-release code (or they can of course use any tag they desire).

yyyc514, 2016-02-29:
It was considered to tag a 'stable' but that really just duplicates the release versions, which are pretty clear in and of themselves.

closed 2016-02-29 by yyyc514

yyyc514, 2016-02-29:
I created a stable branch for now since we only have a single major release series (1.x).

Jezzamonn, 2016-06-15:
Just posting that I'm for the idea of the master being the latest stable release. This seems to be the standard for a lot of projects!

yyyc514, 2016-07-02:
> This seems to be the standard for a lot of projects!

If you say so.  The projects I use (mostly Ruby though) all typically use master for development and then tag releases.  Since we do have release channels (zipfile, Arduino IDE,etc.) I'm not sure the harm in master being development here.

People SHOULD use the development branch and if they want to use stable they should know why and know how to switch branches.  

This is made more annoying though by the fast that for some reason we're scared to death of finally releasing 1.2

MLXXXp, 2016-07-03:
> People SHOULD use the development branch and if they want to use stable they should know why and know how to switch branches.

The problem is, many Arduboy developers are beginners. They find the repository, muddle through figuring out how to grab and install the *master* branch into the IDE (probably not even knowing that other branches exist), and then develop a sketch using it, with reference to the examples, which have been updated to work with the new API. They then make the code available and announce it on the forum.

Next, users who have followed the Quick Start Guide, and have installed the released version of the library, and have no coding experience whatsoever, start posting things like: *I get errors when I try to compile your sketch:*
```
sketch/Utils.h:4:24: fatal error: ab_printer.h: No such file or directory
#include "ab_printer.h"
^
compilation terminated.
```
This has happened numerous times in the forum.

You expect everyone to be an expert or stay away. That's not the philosophy of the Arduboy world. We try to make things as idiot proof as possible.