Ar-zz-duboy/Arduboy #128: IDE 1.0.5 Compatibility Questions

yyyc514, 2016-03-02:
Bring it on, baby!

yyyc514, 2016-03-03:
This is created mostly as a place the codebender people can ask questions - since I believe they are the only ones using such an old platform. :-)

But I have a question: Don't all the files have to go back in a single directory to support 1.0.5, or is that not true?

yyyc514, 2016-03-04:
@dastergon What is the ETA on 1.6 compatibility for Codebender?

yyyc514, 2016-03-04:
For now a codebender branch has been created.  Code bender can maintain their own branch separately and it can be merged upstream to our `codebender` branch, though now I'm being told that the codebender toolchain isn't EXACTLY the 1.0.5 toolchain either, so this is really all about code bender, not exactly about 1.0.5 support.

tzikis, 2016-03-04:
@yyyc514 codebender founder here

we have now officially stopped trying to get your stuff compatible with us. this is not our job, and getting this library to work on codebender was a courtesy to @bateske, who wants to use codebender as the main tool for programming Arduboy, but couldn't develop the changes himself

however, we have spent many hours on this so far, as well as server time to test the library, and many hours on skype calls with @bateske. In return, we have received nothing but spite from the library's developers

Just to be clear. It is not our job to make Arduboy's library work on codebender, it should be working there in the first place, and it's Arduboy's job to uphold their part of our deal. I've already told @bateske, if the master branch doesn't work with codebender by next Friday, I'll withdraw our partnership and explain to our users why that is.

I would prefer it any other way, but we have spent resources we can't afford, for something we'll never profit from, just for the love of the product. And all we got is our brand tarnished on the forum by the state of Arduboy's libraries and integration with codebender. Truthfully, we don't need this, we have other things to worry about

MLXXXp, 2016-03-04:
>I would prefer it any other way, but we have spent resources we can't afford, for something we'll never profit from, just for the love of the product.

You profit (I assume) from getting as many users as possible using your site, which claims to be an Arduino compatible "playground". If people can't use products, sketches and libraries that have been developed for a reasonably current version of the Arduino IDE, then you *do* stand to loose profit.

You have to decide if the effort it takes to support the Arduboy, or the [Arduino Yun](http://feedback.codebender.cc/forums/280540-board-requests/suggestions/6930174-support-for-arduino-yun) or the [Arduino Due](http://feedback.codebender.cc/forums/280540-board-requests/suggestions/6205527-i-don-t-see-support-for-arduino-due) or the [Arduino Zero pro](http://feedback.codebender.cc/forums/280540-board-requests/suggestions/7176562-support-for-the-new-arduino-zero-pro) or the [Teensy](http://feedback.codebender.cc/forums/280540-board-requests/suggestions/7918566-please-add-teensy-3-1-to-the-supported-boards) or the [Arduino Esplora TFT Screen](http://feedback.codebender.cc/forums/165703-general-requests/suggestions/6450533-arduino-esplora-tft-screen-horizontal-bars-fix) or the [Adafruit FONA Library](http://feedback.codebender.cc/forums/280538-library-requests/suggestions/9324006-adafruit-fona-library), etc., etc., is worth the additional profit potential.

MLXXXp, 2016-03-04:
>I've already told @bateske, if the master branch doesn't work with codebender by next Friday, I'll withdraw our partnership and explain to our users why that is.

Does forking to a separate repository, who's *master* branch works with *codebender*, satisfy that requirement?

tzikis, 2016-03-04:
No, it doesn't. We're not kidding anyone here by making a new repo and calling a branch master.

It's simple. If Arduboy's production version of the library doesn't work with codebender, we are dropping support. We've already put too much effort on our own to help with that (which was not our part to do anyway).

Arduino 1.6 is coming to codebender in its time. As for our profit concerns, we support 95% of the boards and libraries out there, so I'm not losing my sleep over that. And no, I don't ever expect to make any profit from our integration with Arduboy. In fact we'll go in $2k most probably as it is

davidperrenoud, 2016-03-04:
Setting the 1.0/1.6 issues aside, the codebender branch seems to work fine on Codebender, with the ArduBreakout example and the Arduboy 1.0. Could someone test it with Arduboy Dev kit?

The problem I see is that all the games will need to be modified to at least have #include <SPI.h> and #include <EEPROM.h>.

yyyc514, 2016-03-04:
@tzikis It's not spite.  It's that none of us (those doing the most contribution right now) care about codebender (no offense).  I think most devs (so it seems so far) are going to use the official toolchain (or a custom toolchain even ) and more importantly *most users* are going to download binaries of the games and install the .hex files directly.  That's coming together really fast.  And there are VERY REAL costs to supporting your old toolchain (esp. when we aren't using your platform).  It slows down development of the library and makes it harder to test changes.

There are also a lot of 3rd party libraries cropping up for Arduboy that right now don't care about your platform at all - so it's really no good if ONLY the core library works but half the games don't compile on your platform because the other libraries user are 1.6 compatible.

And it seems now you want US to pay those costs (you  have no interest in paying them now - you were only doing it as a courtesy) and it sounds like you definitely don't want to continue to keep making the library compatible as the majority of us (on 1.6) continue to improve it.  I would say that you've validated my biggest concern - because I see that causing a lot of potential issues in the future.

A lot off effort already went into upgrading us to 1.6 before anyone even mentioned "hey this MUST work with codebender".  I feel bad for @bateske and whatever deal you guys made that is now stressing everyone out.  And of course when your platform upgrades to 1.6 this all should work just fine.

I'm not 100% against these changes getting into master though at this point the biggest issue is the directory structure.  That feels like a large step backwards to me now that we've already moved forward.  That and ongoing maintenance.  The more "targets" we have the more work for developers and testers... so if we have no one on our team who is a super supporter of Codebender and also a core committer her, then it may not be practical for us to support Codebender.  

We honestly just don't know - which is why I wasn't in a BIG hurry to suddenly REVERT our codebase to 1.0.5 when we've already left it... lets first get a branch working and see what happens with the whole Codebender thing.

> Setting the 1.0/1.6 issues aside, the codebender branch seems to work fine on Codebender, with the ArduBreakout example and the Arduboy 1.0.

How do you test for Devkit? Does Codebender have compile choices?  If so perhaps we could add a "stub" to the library or something that would have just codebender junk it in and compile it conditionally only on codebender.

> We're not kidding anyone here by making a new repo and calling a branch master.

That wasn't my suggestion, just a diff way to make it easier to find.

yyyc514, 2016-03-04:
@tzikis To clear up a few things:

- Are you even using a toolchain identical to 1.0.5 (or exactly)?  Perhaps I've been misinformed but @MLXXXp said he tried some of the things you guys are changing on IDE 1.0.5 and they compiled fine.  It's obviously not workable if we have no way to test new changes against your platform quickly and easily.
- I was told there is no way to update the library on codebender.  Is that true?  That the core github contributors would have to ask permission fro someone at codebender to update the library?
- Does codebender support library versioning?  How do you deal with library upgrades and games that are compiled against different versions of a library?

yyyc514, 2016-03-04:
>  was a courtesy to @bateske, who wants to use codebender as the main tool for programming Arduboy, but couldn't develop the changes himself

Hard to get the meaning here from phrasing.  I don't think @bateske can really decide what the main programing tool is... (that's how I first read it).  That may have been the vision originally, but a lot has happened since then better tools, better binary installers, the library already being upgraded to 1.6, etc.  He may like codebender and use it himself or even use his "official" voice to say it's the official platform or some such, but unless you get those passionately developing the library (and games) onboard that's really all just posturing.

Or you get a bunch of users on board and then use sheer mass to shove it down the throats of the developers. :)  That doesn't solve the problem with 3rd party libraries though.

I'm sure getting 1.6 working is no walk in the park, but it is the solution that would work for ALL involved.  I hope you guys are truly working on that.

MLXXXp, 2016-03-04:
>>Does forking to a separate repository, who's master branch works with codebender, satisfy that requirement?

>No, it doesn't. We're not kidding anyone here by making a new repo and calling a branch master.

What I meant was, [Arduboy](https://github.com/Arduboy) takes what's currently in the *codebender* branch of [this repository](https://github.com/Arduboy/Arduboy) and itself creates an entirely new repository: *Arduboy/ArduboyCodebender*, who's **_master_** branch is fully codebender compatible.

The Arduboy organisation would maintain this new repository in parallel with [*Arduboy/Arduboy*](https://github.com/Arduboy/Arduboy), as best as possible, and see how it goes.

codebender just has to pull and install from *Arduboy/ArduboyCodebender* to get a codebender compatible version of the library. No more effort is required from codebender, other than letting us know (and possibly offer advice on fixes) if a new version of *Arduboy/ArduboyCodebender* has issues with installation or operation, which the Arduboy library maintainers would try to address in a timely manner.

MLXXXp, 2016-03-04:
>How do you test for Devkit? Does Codebender have compile choices?

I expect that we would have to supply a *boards.txt* file, or equivalent, that codebender would use to add *Arduboy* and *Arduboy DevKit* to their supported boards.

Since I don't think codebender supports extra board selection submenus, treating the production Arduboy and the DevKit as separate boards would probably be a good idea, as I suggested [here](https://github.com/yyyc514/hardware_arduboy/pull/1#issuecomment-189554307)

MLXXXp, 2016-03-04:
@davidperrenoud, you wrote:
>the codebender branch seems to work fine on Codebender, with the ArduBreakout example and the Arduboy 1.0.

Did you have to do anything with the [ArduboyPlaytune library](https://github.com/Arduboy/ArduboyPlayTune), used by ArduBreakout? ArduboyPlaytune is currently in IDE 1.6 format.

yyyc514, 2016-03-04:
>  which the Arduboy library maintainers would try to address in a timely manner.

I'm just curious which maintainers you are referring to.  Not a lot of people contributing time to the library so far.

yyyc514, 2016-03-04:
> I expect that we would have to supply a boards.txt file, or equivalent, that codebender would use to add Arduboy and Arduboy DevKit to their supported boards.

Very interested to heard of codebender can set arbitrary compiler defines, that would be interesting.

MLXXXp, 2016-03-04:
Maybe not *arbitrary* compiler defines, but if we treated each board as separate, the *...build.board=* parameters in *boards.txt* would be passed to the compiler as a define, which is what we would use instead of *ARDUBOY_10* and *AB_DEVKIT*.

If *boards.txt* contained  `arduboy.build.board=AVR_ARDUBOY` for production and `arduboy.build.board=AVR_ABOY_DEVKIT` for the DevKit, the compiler would be passed `ARDUINO_AVR_ARDUBOY` or `ARDUINO_AVR_ABOY_DEVKIT`

This would work for both the 1.6 compatible library and the codebender version.

MLXXXp, 2016-03-04:
>I'm just curious which maintainers you are referring to. Not a lot of people contributing time to the library so far.

I assumed that porting changes from the main version to the codebender version wouldn't take much time, as long as both libraries were close to identical. I'm also assuming that the flurry of changes currently taking place will subside and relative stability will ensue. That's why I said _"[...]as best as possible, and see how it goes"_. And again, we have to trust that codebender will eventually become 1.6 compatible (hopefully soon).

Will a PR merge automatically if the changed file resides in a different directory in the repository? Git has heuristics to somewhat be aware that a file has been moved.

yyyc514, 2016-03-04:
> If boards.txt contained arduboy.build.board=AVR_ARDUBOY for production and

Does that platform even have a boards.txt?  Lots of unanswered questions.  It would be very interested to know if one can target the Codebender platform with custom code based on compiler defines.

> I assumed that porting changes from the main version to the codebender version wouldn't take much time

Maybe, maybe not.

> Git has heuristics to somewhat be aware that a file has been moved.

It does, but honestly I don't have much experience with a fork with files that have been moved to entirely different locations.  I imagine one would find out pretty quickly how easy or difficult it is.

MLXXXp, 2016-03-04:
>Does that platform even have a boards.txt?

I don't know, but it would certainly need to have the equivalent, at least to provide *F_CPU*.

However, it looks like *..build.board=* is new to 1.6, so that could be a problem.

yyyc514, 2016-03-05:
@tzikis @dastergon

So we have a working branch. @davidperrenoud says it works fine on Codebender - though I'm unsure what that means because I thought you guys had to "install it manually" with some magic sauce as an actual library before we could say for certain it 100% works.  There is also a question as to whether it will work with both production and dev kit Arduboys.

I think (on this end) we're curious to see how much effort maintaining the 105/codebender branch (now that we have one) will entail... whether that is a small amount of effort that myself and current contributors can do easily or whether that will be a larger effort that will require finding additional contributors that happen to be passionate about the Codebender platform to do that work.

Right now what I hear from @bateske and friends seems to be a huge push to "get it working on codebender".  To read further into that than it's face value is just politics.  I don't think there is anything further we can do on this end to accomplish that goal.  If there is, please ask questions or let us know.

So to summarize:

- We have viable potential Codebender compatible code that you guys helped us with.
- It's in the `codebender` branch - https://github.com/Arduboy/Arduboy/tree/codebender
- Does that code indeed work on Codebender?
- Does it work with the DevKit and Production Arduboy both, or will only Production be supported?
- If not, what further code changes might be necessary (regardless of who will do the work)?
- If the code is functional, then we should work on a joint release of Arduboy library 1.2 both on Github and on Codebender.  (currently you're PRs were all against a pre-release branch)
- How long does it take you guys to setup the code as an official library and Arduboy as an official platform?
- Can the Arduboy library maintainers update the library on Codebender as we improve it and release new versions, or can only Codebender staff do that?

@tzikis @dastergon - Balls in your court!

MLXXXp, 2016-03-05:
>I thought you guys had to "install it manually" with some magic sauce as an actual library before we could say for certain it 100% works.

On codebender, if you want to use or test an updated or unavailable library, you can [embed it in the sketch](http://feedback.codebender.cc/knowledgebase/articles/92166-embedding-libraries-inside-a-sketch) (just like you can with the Arduino IDE) or install it as a [personal library](http://feedback.codebender.cc/knowledgebase/articles/399209-using-personal-libraries).

MLXXXp, 2016-03-05:
>How does library versioning work in general on your platform?

Not official, but my understanding is they maintain one version of a given library and keep it up to date manually, buy pulling the code from the official location, either occasionally or upon request.

tzikis, 2016-03-07:
Apologies for the delay, but I'm traveling at the moment. It's nice to see some actual questions that help move things forward instead of pure resistance, though. I've tried to address all comments and questions below

## Re: Arduboy and codebender
I've made a partnership with Arduboy, and based on that I've made a promise to my users as has Kevin to his customers. That promise is that Arduboy and codebender will work together to ensure Arduboy is supported on codebender at all times, the drivers taken care of and the library maintained and updated. If you guys insist that Kevin doesn't get to decide, then my withdrawal from our partnership still stands, as I won't be able to keep my promise to my users, given that I don't have any assurances that Arduboy will keep its end of the bargain. From what I understand it has been made clear that Kevin is still the lead of this library and decides the direction it takes and codebender is officially supported and maintained.

If that's not so, or that's not clear, then I would like to know (and I'm entitled to know) before proceeding any further.

## Re: Arduboy library & compatibility with codebender
Again, I want to be ***perfectly*** clear. The only solution we will accept is using the master branch of the Arduboy library, developed while taking codebender into account. As I mentioned before, we won't use a second branch that "you'll try to update and see how it goes". We also expect that the library has been tested with either the Arduino 1.0.5 IDE or codebender (which is as hard as uploading the library on your personal account and trying it out) when we get a request to update it

The way library requests work is that you (or anyone) make a request on our feedback platform. We then test the library by compiling all projects that use it to make sure nothing breaks. If there's anything wrong, we notify the developers. If everything works fine we update it and notify all people who use it.

There are certain cases where the library maintainers want to go ahead with versions that break a small amount of sketches. If that happens we go ahead but notify the people affected by it as to how to fix their projects.

Having said that, please keep in mind that this does not absolve the developers of the responsibility to test the library with codebender before issuing a request. It's not our job to keep the library compatible with codebender, it is Arduboy's. We merely have our own checks to ensure high quality and double-check for issues, in a way only we can (since we have access to the thousands of sketches on codebender).

So yes, we do have one version of the library at any give time, that everyone uses, and we make sure that works perfectly for everyone. And yes, you do have to ask us to update it, for good reason. It's to provide a great level of support to our users, not because we want you to "ask permission".

This is also why you want to avoid using the library as a personal library, and/or bundling it in the project itself, except for testing purposes.

As for 1.0.5 and codebender's compatibility, we use the exact same toolchain. There are a couple of edge cases because this thing runs on the cloud, but they don't matter for 99% of the time.

## Re: Arduboy and Arduboy DevKit
We do use boards.txt files for all our boards. However, as MLXXXp mentioned, you can't use arduboy.build.board on 1.0.x. I've already told Kevin a way to make both work on IDE 1.6.x and codebender, but it's not clear he wants to continue supporting the DevKit because it's EOL

For the record, my suggestion is to get 2 USB PID/VIDs from the OpenMoko foundation, and use a different one for each board. The way the Arduino IDE works with PID/VIDs is that they are passed as -D options on the compiler, so the library can detect which board it's being compiled against with #ifdefs.

This is a decision beyond me, so I won't get any more into that. For now I'm assuming that we'll only be supporting the Kickstarter Arduboy because that's what I know

## Next Steps Forward
As I mentioned above, we are not going to accept a library update request other than from the master branch of the actively developed library. Re-compiling and testing all the sketches is a seriously expensive and time-consuming process.

Therefore, we expect you to try the branch on 1.0.5 and/or codebender (compiling all bundled examples is always a good idea), and make any necessary changes. The end goal is a working, and tested, master branch that works with codebender for all intents and purposes, and tested to the extent of your abilities. This is what we will be looking for on updated versions of the library as well, of course.

After that, we can go ahead and test the libray. Assuming it doesn't break anything, it should be added/updated within a couple of days (usually on Tuesdays/Wednesdays)

Thanks for your time, I know it was a long reply

yyyc514, 2016-03-07:
> If you guys insist that Kevin doesn't get to decide, then my withdrawal from our partnership still stands, as I won't be able to keep my promise to my users, given that I don't have any assurances [...]

I think there is a disconnect here.   You use the term "Arduboy" like it's a singular thing or entity or person.  And while it's true Kevin is probably the "face" of Arduboy he isn't doing any of the major improvements on the library.  He isn't doing any of the major work on hex loaders, web editors, or even whole alternatives to Codebender.  He's built an amazing device, for sure - and we wouldn't be here without him, for sure - but Arduboy is a lot bigger than Kevin now.  Kevin is great and I hope he sticks around and continues to develop the hardware, but he can't do it all.  Not enough hours in the day for one. :)

Passionate volunteers (including myself) have jumped in with our own ideas and we've improved the library significantly since the early days.  Arduboy has benefited from this.  It will no doubt continue to benefit from this.   But other than freezing the library and saying "it's done" I'm not sure how Kevin can guarantee to keep this promise to you long-term.  **The library long-term needs core developers who care about Codebender and Kevin's promise to support Codebender.**

If those people show up then the problem takes care of itself.  *If those people do not show up, then I think you cannot have what you want.*  I think it is very short-sighted of you to now *demand* us to do something now (merge a specific PR to master) that the current core contributors are against.  A few people (no names) even suggested we "just give you what you right now"... get you happy, get the library on Codebender then in 2 weeks we can just revert those changes, put master back where we want it, and put you right back in a branch.

You seem like a smart guy.  Hopefully you can see how being in master isn't any magic guarantee of our future support - but that's really true for any OSS project.  Any time in the future it could be decided to rip out support for 105.  Maybe 1.7 comes along and offers some amazing improvements... boom, suddenly 105 is out again and 1.7 is in.  I suppose if Kevin owns `Arduboy/Arduboy` he can be a dictator there and say "We will live and die on 105, period" - but when the majority decides otherwise the true library just moves on and lives in someone else's Github repository - and no one (except Codebender) uses the "official" version.

We (those actively involved in the libraries development) already made the decision (after much discussion) a while back to move on to the 1.6 format, and personally I'm not one to move backwards... I might feel a little differently if we were still on 105 and hadn't moved the files around.  The 1.0 platform is very old now.  It was first released in 2011.  105 itself is 3 years old; the 1.5 branch dates back to 2012; 1.6 was released over a year ago.  105 is only getting older every day.  Yes, I realize it's just shuffling files around, added additional headers, etc, etc... but it all adds up.

AND: 
Supporting a whole platform (like Codebender) is a large undertaking - you've made that VERY clear in this thread how much work it is and how you guys are unwilling to carry that burden yourselves.  I'm not sure we realistically have the developers for that currently.  I know I don't have the extra time to always be checking everything I do against Codebender (or any interest in installing Java and a 1.0.5 toolchain). I work on the library in my free time because I'm passionate about it.   I'd rather have a vibrant library that only works on 1.6 than a stagnant library that ALSO works on Codebender.

A lot of work is going on to replace [the need for] Codebender, so I'd be more likely to bet on those projects (as they aren't stuck on 105) going forward.  So far I've heard no dates from when Codebender plans to support 1.6.  I'm also on record as saying I think the future is .hex files (for simple game distribution).  

To summarize:

I think until we see that there is REAL grassroots support for Codebender in our developer community that a separate branch makes the most sense for everyone.  It gives us a version of the library that is guaranteed to work and compiles on Codebender, and it also prevents the development effort from slowing down or being burdened by 105.   I don't think anyone can "promise" you that that branch will be kept up-to-date.  *I was at least sincere in my statement of seeing what that efforts might look like long-term.*

Also (as stated above) getting one particular PR into master is no guarantee that we're going to support Codebender (or 105) long-term (or even short-term) as development continues.   Long-term either developers who love Codebender will step up to do that work, or it might be Codebender ends up with a static version of the library - and that is true regardless of what branches are named.  Earlier I mentioned someone suggesting we just rip out 105 out a few weeks later, but one could also imagine us just forking a 'nextgen' branch and doing all future development work there...

*waving hand like a Jedi* `master`, this isn't the branch you're looking for.

I feel bad for you and Kevin and the promises that were made that led to where we are now, but it is what it is.  I have nothing against Codebender (or you) personally.  Please take the time to consider my points above and let us know if we can move forward with getting the codebender branch on Codebender and let the future worry about itself - as it will do in any case.

tzikis, 2016-03-08:
Well in that case, my comment stands. We're not supporting Arduboy unless we get the same support back

I would consider it mistaken that the developers of this library get to decide what Arduboy works or doesn't work with officially. That's something only Kevin can do, he's the one who sells the device, and decides what software the Arduboy works with. If we wants to hear you out or give you the power to do so, that's his prerogative. Just as anyone else is free to do whatever he wants that would work with Arduboy, but that doesn't make it official and supported

So again, and just to ping @bateske to this, if we can't be guaranteed that Arduboy will keep their end of the bargain and have a library request for a working, compatible, and actively maintained Arduboy library, we are withdrawing from this partnership.

yyyc514, 2016-03-08:
> I would consider it mistaken that the developers of this library get to decide what Arduboy works or doesn't work with officially. That's something only Kevin can do, he's the one who sells the device, and decides what software the Arduboy works with.

I honestly don't think you can avoid that in a truly OSS project.  It's a little more nuanced when you throw in the word "officially", but my major point still stands.  If none of the active developers have the time or interest to support Codebender, it won't be supported. I can speak only for myself (directly), but the few others actively contributing also haven't been particularly interested in Codebender.  So unless *someone* wants to step up and do the work... it's not going to get done.

Find someone to step up and do the work, and it will get done.  It's a really simple equation.  I un-merged the PR the other day because at the time it was marked "Please Hold" and should not have been merged as quickly as it was.  I didn't feel it had been discussed long enough or that anyone had considered the long-term consequences.  I think we've more than had that discussion now, even if I've failed to be persuasive.

I'll step back now and let whoever else do whatever they will.   I think you probably have more than enough political pressure to get your previous PR merged back into master, if that's still your demand.  It just won't be me merging (or testing) it;  **but again, *please* do not naively assume that being "officially" supported is the same thing as being actually supported.** 

I don't say that in a mean way, just an honest way.

@rogosher Guess the ball is in your and @bateske court again.  I suggest you try to find some developers who are truly passionate about the Codebender platform and 1.0.5. :-)

bateske, 2016-03-11:
@yyyc514 If you don't want to do it that's fine. Your attitude is not helping anyone here. We will manage these pull requests yourself.

You've clearly got an axe to grind here and frankly it's not appreciated. We are looking for team players and while we really appreciate your feedback and contributions... at the end of the day this is a business and needs to be treated as such.

I think you forget that the point of doing all of this is to make it easier for beginners to get started. Following the current use case right now is dramatically more steps than using codebender.

Lastly, your argument for nobody supporting codebender is one hell of a catch 22. Of course nobody uses it, the library doesn't work and there is nobody directing users there.

@rogosher if you can please make the updates needed, that would be great. I'll give you the recognition of getting it done.

MLXXXp, 2016-03-11:
>I think you forget that the point of doing all of this is to make it easier for beginners to get started.

It may be making things a bit easier for beginners, but it's also tying their hands somewhat by using an obsolete development environment which will lack some functionality and may not match the current Arduino documentation, thus possibly causing confusion.

rogosher, 2016-03-11:
Make it work once, and forget about it. If the codebender IDE is ever update to 1.6 then thats cool. Otherwise, once a stable branch is done, it's done forever.

Anyway, for sure, now that the dust has settled I will pick up the task of getting Codebender and Arduboy functioning.

yyyc514, 2016-03-19:
> You've clearly got an axe to grind here and frankly it's not appreciated. We are looking for team players and while we really appreciate your feedback and contributions... at the end of the day this is a business and needs to be treated as such.

**For you it's a business.**  For me (and many others) it's just a cool Kickstarter project and a fun _open-source_ project I've been happy to contribute to - because it was fun for me and I get to work on things that interest me and learn things I want to learn.  That's why a lot of others contribute, also.  Don't presume everyone who takes an interest and contributes shares all your business objectives.

I've always cared more about the _library_. Codebender may indeed be good for the _project_.   I don't think it is good for the core library though (and I and others have already clearly stated why), hence why I created a separate branch. **In my opinion that's still the way to have the best of both worlds:**  

1. Codebender has a library now to support Arduboy (their own branch)
2. The core library is not limited long-term by the limitations of the obsolete 1.0x build environment.

_It's unfortunate Codebender doesn't agree._

It's funny now that I think about it because we're actually trying to achieve the same things here, but just thru vastly different means. I honestly think _everyone_ is better in the long-term with a branch for Codebender.  I think it would be irresponsible for me NOT to make this point loudly and clearly.

> Lastly, your argument for nobody supporting codebender is one hell of a catch 22. Of course nobody uses it, the library doesn't work and there is nobody directing users there.

Not what I said at all.  I said so far it seems the most prolific contributors to the library don't care that much about Codebender.

> Otherwise, once a stable branch is done, it's done forever.

That's only true if you stop updating the library... otherwise long-term maintenance is required to test every release on Codebender to make sure no issues have been introduced. Or maybe you're trying to say a lot with that one word "stable".

I already created a branch (with Codebender's own changes) that supposedly works with Codebender, and Codebender was not happy with that.  If "once a stable branch is done, it's done forever." was true you think they'd have been happy with that as a start.  I mean it was 1.2, but if we released 1.2 then that would of course become a "stable" branch.

> It may be making things a bit easier for beginners, but it's also tying their hands somewhat by using an obsolete development environment

Thanks for all your library contributions and making sure I'm not a solo voice on this issue! :-)

I'm still very curious to see how all this will play out.

rogosher, 2016-03-26:
See https://github.com/Arduboy/ArduboyClassic

closed 2016-03-26 by rogosher