Ar-zz-duboy/Arduboy #131: 8Mhz battery life vs 16Mhz

yyyc514, 2016-03-04:
Anyone want to compile a 8mhz sketch that just does a constant animation (without sleep!, so you can't use the nextFrame() API) and compare it to a 16Mhz sketch and see what the battery life is with both?  I think it would be super helpful to know what the difference is before we decide how much time it's worth pouring into making things work at 8Mhz.

If it's 10 vs 20 hours of battery life I think you could argue it's wroth it for some games.  If it's 10 vs 11 hours the argument is a lot weaker.

MLXXXp, 2016-03-04:
I currently don't have the time do this, and not having a real Arduboy may invalidate my results, anyway.

**Note** for anyone wishing to try this:
Using *F_CPU* to change to 8MHz will make the USB interface stop working. You'll have to use the reset switch or the *hold LEFT/UP buttons* technique to upload a new sketch.

yyyc514, 2018-08-31:
No one seems to care about this.  I will close.

closed 2018-08-31 by yyyc514