Ar-zz-duboy/Arduboy #136: drawBitmap shifting everything to the right?

amokan, 2016-04-28:
Seems like something is wrong with `drawBitmap`, at least on an 8x8 bitmap. Will dig a bit more, but just wondering if I'm crazy...

rogosher, 2016-04-28:
It is most likely in how you are packing bits. Each byte in the array holds the pixel value for 8 cells. That is each bit in a byte will be a pixel on a screen.

searching the http://community.arduboy.com page will reveal a lot of similar discussions on this, and some algorithms for packing and unpacking.

I'll leave this issue open, as it is worth checking to see if you are on the 1.1 tagged release and not the master. If the behavior from the 1.1 tag differs from the master, then yes an error has been introduced in that function, but I do not think it's been poked at much lately.

yyyc514, 2016-04-28:
Draw a bitmap that is all white and see if it's still "off".  If not then it's likely your bit packing as rogosher already said.

rogosher, 2016-05-27:
Working under the assumption the issue was the byte packing for the screen. Thanks for opening an issue : )

closed 2016-05-27 by rogosher