Ar-zz-duboy/Arduboy #139: License is BSD but Playtunes code is GPL v3

MLXXXp, 2016-06-10:
I am not a lawyer, but I believe that since this library integrates code from [arduino-playtune](https://github.com/LenShustek/arduino-playtune), which is licensed under the GNU General Public License v3, this library can't use the BSD license.

Since it uses GNU GPL v3 code, this library should be licensed under GNU GPL v3.

yyyc514, 2016-06-10:
That code has been ripped out now into a separate repository - and the original license was also always mentioned in the contributions file.

yyyc514, 2016-06-10:
I'll let someone else decide if any other changes are necessary here... but if so that issue probably should be opened against the other repo.

MLXXXp, 2016-06-10:
Yes, I realised after opening this issue that moving the Playtune stuff to a separate repository eliminated all GPL code, so going forward this repository could be BSD licensed. However, at the moment the stable V1.1 code that people are installing still includes Playtune code. Until we release a version without Playtune code, we should be using GPL v3.

GPL is a "viral" license. If you incorporate any GPL code in your source, you must use GPL as the license for all of it. You can't just acknowledge that portions of your code are GPL and have the rest remain BSD.

The ArduboyPlayTune repository currently contains no license whatsoever. [The PR that I submitted](https://github.com/Arduboy/ArduboyPlayTune/pull/6) 2 days ago adds the GPL v3 license.

yyyc514, 2016-06-10:
Someone else will have to sort this out.  The library was largely rewritten after we pulled it in... and really it should be treated as vendored code within our project (even in 1.x)... we should add the GPL license notice to the tunes source files only.

MLXXXp, 2016-06-10:
> The library was largely rewritten after we pulled it in... and really it should be treated as vendored code within our project (even in 1.x)

Not if you're going to acknowledge where it came from in the *CONTRIBUTORS.md* file. You could leave it as BSD if you wanted to pretend it was written from scratch.

> we should add the GPL license notice to the tunes source files only.

Which, as I said, I did in the PR.

closed 2016-08-31 by rogosher