Ar-zz-duboy/Arduboy #142: Documentation

BenMcLean, 2016-06-18:
I understand that Doxygen is supposed to auto-generate the documentation from comments in the source code. But to be helpful to all the newbie coders, this info really ought to be on the web somewhere. Maybe I'm just spoiled by Javadocs but this would be nice

rogosher, 2016-06-23:
++ 
I was thinking that Github does have a webspace available to projects, but I don't believe it will run php as needed. There have been a few servers up now and then that hold them, but the official one will probably be hosted at something like docs.arduboy.com? Suggestions?

MLXXXp, 2016-06-23:
> docs.arduboy.com? Suggestions?

Why create a *docs* subdomain? What's wrong with a directory under the main domain?
*arduboy.com/docs/*