Ar-zz-duboy/Arduboy #143: Document units such as duration

joho, 2016-07-02:
`void ArduboyTunes::tone(unsigned int frequency, unsigned long duration)`

I've no idea what unit I should be specifying duration in. Seconds? Milliseconds? Would be great if those could be included in comments in the header, or even in the argument name `durationInMs` or something.

MLXXXp, 2016-07-02:
I guess since it's meant to replace [Arduino tone() ](https://www.arduino.cc/en/Reference/Tone) it was assumed that it would be understood that it's in milliseconds.

You're point is noted, though.

joho, 2016-07-02:
Right. Sorry, I had no idea. Thanks. Arduboy's my first foray into arduino, there's a little bit of a learning curve :)

yyyc514, 2016-07-02:
durationInMs isn't the worst idea

closed 2018-02-03 by joho