Ar-zz-duboy/Arduboy #146: Should the new draw() function be simplified and renamed?

MLXXXp, 2016-07-18:
I don't really see much of a point for having the new *draw()* function, which was added in the *develop* branch. It just saves you from directly calling *clear()* and *display()*. As with *nextFrame()*, you still have to call it continually until it indicates that it's time for the next frame. I think it would be better if it waited until the next frame time before returning, and I don't think it needs to return any kind of status.

Also, all the other functions that are prefixed with *draw* only manipulate the screen buffer in RAM. They don't actually change what's displayed on the screen itself. Therefore, I think the name *draw()* is confusing. Perhaps *render()* or *renderFrame()* would be more meaningful. Using the name *renderFrame()* makes it clearer that it is part of the timed *frame* functions.

I propose that the function be:

```cpp
void ArduboyBase::renderFrame(void (*f)())
{
  while (!newFrame()) { } //(!nextFrame() if my "newFrame()" pull request isn't accepted)
  clear();
  (*f)(); // call the function passed as paramter to draw
  display();
}
```

A sketch might then contain

```cpp
void setup()
{
  arduboy.begin();
  arduboy.setFrameRate(30);
}

void loop()
{
  arduboy.renderFrame(gameFrame);
}

void gameFrame()
{
  // code to update the screen buffer with the next frame contents
}
```

rogosher, 2016-08-06:
I agree, and in the develop  a `perform()` function that can be passed a pointer to a function has been included. 

The notion of a "draw" routine, the original name, that performs anything from print to poll input is probably the wrong approach. However, a set of handlers for this sort of thing, like a `draw`, `event`, `joy`, etc. is plausible.

The verb _perform_ seems to capture the idea, that is: "Arduboy, perform _this_", where this is a routine (keeping in mind `do` is a reserved keyword :cry:), but another verb might work better. 

It would be nice to do everything that "begin" does in a constructor perhaps, that would reduce the number of calls a user is required to make.

yyyc514, 2017-04-19:
Think this was the right call. :)

MLXXXp, 2017-04-19:
@yyyc514 You can close this issue if you like
(you're still a collaborator, aren't you?)

closed 2017-04-19 by yyyc514